Xilinx/Vitis-AI #1003: Does the current second-stage network support your platform?

jia0511, 2022-09-07:
Hi  xilinx,

At present, we have a two-stage network, and cascade rcnn wants to be deployed on vck5000, but I only see a one-stage network in modelzoo. I want to ask whether the quantization and compilation of rcnn network is currently supported?

jia0511, 2022-09-13:
Hi xilinx,

Can you give us an answer, whether the current two-stage model can be quantified normally using the vitis-ai tool？

SuZhan0322, 2022-09-13:
Hi @jia0511,
Vitis-AI tools do not support two-stage models yet, thanks~

jia0511, 2022-09-13:
@SuZhan0322 

Is there any plan for support in the future, and how can we support it?

SuZhan0322, 2022-09-13:
Hi @jia0511 ,
It is planned to support this feature in VAI3.0 WeGo flow. thanks!

jia0511, 2022-09-13:
@SuZhan0322 

Thank you for your support！

closed 2022-09-13 by jia0511