Xilinx/Vitis-AI #1005: Steps to quantize, compile and run inference a yolov4 darknet model on ZCU102?

vongracia, 2022-09-07:
Hi all,

I'm trying to make inference on a ZCU102 using a yolov4 darknet model trained on custom 4K data, but I'm having issues. It is explained all here: https://github.com/Xilinx/Vitis-AI/issues/1002

Could anyone provide me a very good tutorial with code to achieve so? I've tried:
https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.3/Design_Tutorials/07-yolov4-tutorial

But I have problems.
Thanks in advance,

qianglin-xlnx, 2022-09-08:
Hi @vongracia  
Which version of Vitis-AI do you use?
I see there is update version of 07-yolov4-tutorial, which is 1.4. Maybe you can try this: https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.4/Design_Tutorials/07-yolov4-tutorial

vongracia, 2022-09-11:
Yes, thank you. I will use that from now on

zhenzhen-AMD, 2022-09-23:
Hi @vongracia ，

Through analysis, the model contains "softplus",
Currently DPU does not support "softplus", so please replace "softplus" with other ops.

[image]

vongracia, 2022-09-28:
Thanks @zhenzhen-AMD 

Do you mean when training the yolov4 using darknet? You mean replacing it by Relu?

Thanks

zhenzhen-AMD, 2022-12-28:
Hi @vongracia ,

There are two solutions:
1. When training, you can replace 'softplus' with 'relu'. Eventually 'relu' will run on the DPU.
2. If you keep 'softplus', you need to set "add_shape_info=True" in the quantization stage. When compiling 'softplus' will be treated as 'unknown', and eventually it will run on the CPU.

Regards,
Zhenzhen

closed 2023-01-16 by qianglin-xlnx

zhenzhen-AMD, 2023-01-16:
Hi @vongracia ,
I will closed this issue. Please free open new issue If you have new problem.