Xilinx/Vitis-AI #1009: Problem connecting to ZCU102 - USB UART cable (cannot open /dev/ttyUSB0)

vongracia, 2022-09-11:
Hi all,

I am not able to access my ZCU102 using minicom. The command I use is:
sudo minicom -D /dev/ttyUSB0
**minicom: cannot open /dev/ttyUSB0: No such file or directory**

In the past I used this tutorial and it worked perfectly:
https://www.centennialsoftwaresolutions.com/post/edge-ai-setup-on-zcu102

Suddenly, I had to reset my laptop and now the port /dev/ttyUSB0 is not accessible. Do you know what could be happening?
I'm using Ubuntu 22.04.

Thank you
Antonio

tianfang-fafafa, 2022-09-14:
@vongracia  seems your laptop can't recognise serial port(ttyUSBx),  could you try to change another laptop or machine?
and is your laptop's system a virtual machine ?

vongracia, 2022-09-15:
It is not a virtual machine, it's a laptop with fresh Ubuntu 22.04 installed. @tianfang-fafafa 

Please check this #1013 , the problem derived into something else and I opened to the forum.

Maybe you can give me a hand,
Thank you very much

qianglin-xlnx, 2022-09-26:
refer to https://github.com/Xilinx/Vitis-AI/issues/1013

closed 2022-09-26 by qianglin-xlnx