Xilinx/Vitis-AI #1012: tensorflow vgg 16 can't inference after convert to xmodel

HuiJu1218, 2022-09-15:
Hi,

Here is my quantize.model and xmodel
[quantize model and xmodel](https://drive.google.com/drive/folders/1br27QEFIp94VolTkD4Rwg51NpmXBPVgn?usp=sharing)

my environment:
- vitis-ai 1.4 cpu
- convert model in tensorflow1 container

I had use keras to get vgg 16 pre-trained model and set model like following. 
[image]

[image]
And I had convert the vgg 16 model to xmodel successfully. But when I starting inference with xmodel, I get error message. The message seems like that model get error output layer.
[image]
I have several question. Does anyone can help me?
1.  I have use netron to compare the xmodel from vitis-ai model zoo and the xmodel that I convert form tensorflow. It looks very different.
2.  The error message shows that correct output must be [1, 2], but get output [1, 2048].

[image]

lishixlnx, 2022-09-16:
open your .xmodel, you can see:
[image]

there is a cpu-subgraph in the model, the [1, 2048] is the output of the "download" node point to the cpu subgraph.  the output of the 2nd Dpu-subgraph is [1,2],  which is what you really need.
The key problem is : the model is divided by 1 cpu-subgrpah, this is because there's op that dpu doesn't support. 
from the structure, it's mainly because of the "add" & "mul" op.

Innodisk-Will, 2022-09-20:
The "add" & "mul" op is "Batchnormalization layer" convert to ".pb" get it. Then, I will remove the "Batchnormalization layer" and training, and convert it to ".pb" & ".xmodel" can run it in KV260. This might not be the best solution, but this works for me.
[image]

closed 2022-09-22 by HuiJu1218