Xilinx/Vitis-AI #1013: Accessing the ZCU102 board using the JTAG, instead of UART? Please help.

vongracia, 2022-09-15:
Hi people, I'm experiencing problems access my ZCU102 board from my laptop with Ubuntu 22.04.

The thing is the following:
- 2 weeks ago I was able to connect from my laptop to the ZCU102 board using the J83 **UART port.** I set the board in SW6 boot mode, and I used this minicom command to access: sudo minicom -D /dev/ttyUSB0   . It booted without problems  and I ran inference on the board without problems.
- Now the UART seems not to work anymore. In case it is broken my only salvation to connect to the board is via the other port, the **JTAG**. If I connect the USB cable going from the laptop to this JTAG in the board, ttyUSB0 seems to be recognized. if I run: ls /dev/ttyUSB*, the output is: /dev/ttyUSB0

My conclusion is that the JTAG, but maybe not the UART, is working. My questions now:
- Can I boot my ZCU102 and access it from my laptopt using the JTAG instead of UART port? All the tutorials describing how to access the board use the UART as example, not the JTAG.
- If so, could you provide me hints on how the switches and pins must be set in the board to boot via JTAG, and also possible configuration or parameters for minicom to properly access the board?

This is driving me crazy, I'm in a rush for some deliveries and a help by someone experienced would be extremely amazing
Thank you

tianfang-fafafa, 2022-09-15:
@vongracia hi,

JTAG can be used to load BOOT.BIN, Image and rootfs, but output info will use serial port.
If your laptop machine can't recognise ttyUSBx when connect your laptop machine with serial port(UART) of zcu102 board, you can try to use monitor for output

vongracia, 2022-09-15:
Thank you so much @tianfang-fafafa 
Could you give me a hand or more details about that?

Up to now, I used to connect the ZCU102 via UART and was able to run inference simply:
1- Open an ubuntu terminal
2 - Enter: sudo minicom -D /dev/ttyUSB0 
3 - Then switch on the board, wait 15 seconds for it to boot and then
4 - Then a command line like the following appears: 
**_root@xilinx-zcu102:_**
5 - Using this command line I was able to copy files from my laptop to the board, and run inference directly on the ZCU102

I followed this simple and short tutorial to do so: https://www.centennialsoftwaresolutions.com/post/edge-ai-setup-on-zcu102

So, how this entire process should be done (or replaced) using the JTAG instead? I'm lost at the moment and need some help for the steps required.

Thanks

tianfang-fafafa, 2022-09-15:
@vongracia JTAG can used for loading image, you can refer to petalinux-boot --jtag from ug1144, but I don't think you need use this function which can't display output.
I can use monitor(DP interface) or network to access to board cmdline

vongracia, 2022-09-15:
Thanks @tianfang-fafafa 

I tried to check on the internet further information on specific steps for doing the process, but I was unable due to my little knowledge on the subject. 

Could you provide me with specific website, link or resources that indicate in detail how to achieve accessing the ZCU102 using JTAG and make inference?

Thank you very much

tianfang-fafafa, 2022-09-15:
@vongracia 
sorry, I can't understand why you want to use JTAG.
your SDcard of zcu102 has already contained image for booting. If this is right, you can boot zcu102 board up. And can replace UART with monitor or network.  how to use network, you can refer to https://www.centennialsoftwaresolutions.com/post/edge-ai-setup-on-zcu102;  (first, you should edit network configure file, make board network use static IP,  via SDcard reader， because your laptop can't recognise ttyUSB so that you can't obtain IP from UART)

vongracia, 2022-09-15:
I want to use JTAG port because I cannot use UART anymore, I think the port is broken. 
- The configuration that worked for me until 2 weeks ago is this:
https://ibb.co/mByqYsz

- As **UART is not** working now, I'm trying to use JTAG port instead. Is this assumption incorrect? See image of how I'm trying to overcome the fact that the UART port does not work.
https://ibb.co/kc6Hww9

I cannot follow this tutorial (https://www.centennialsoftwaresolutions.com/post/edge-ai-setup-on-zcu102;) anymore because in the step where it says:
Open up the terminal and view the boot log through your UART connection to J83
sudo minicom -D /dev/ttyUSB0

Now this does not work and cannot boot the board because it assumes using the UART port, that does not work. I was wondering ways to achieve accesing the board using JTAG instead. If there is another way, please let me know.

Thank you very much
Thanks

tianfang-fafafa, 2022-09-15:
@vongracia 
sorry，I still can't understand. There is no cause and effect between booting and UART.
In this linking https://www.centennialsoftwaresolutions.com/post/edge-ai-setup-on-zcu102, please look at "Place the flashed SD card into J100 (see Fig. 1) ".   When you power on, board will boot up soon. Output will disply on monitor (DP interface). You can acccess to board by USB-keyboard which connected to board.

vongracia, 2022-09-15:
Thanks for your answer. I understand what you mean @tianfang-fafafa 

The output is displayed via DP interface on any monitor. I can access the board by connecting USB keyboard and mouse to the board. But my main question is then how can I then transfer files (the trained models to make inference) from my laptop to the flashed SD card inserted on the board. 

I thought this must be primarily done connecting the laptop to the board (using ethernet cable), and then running minicom to access the board. The problem is that minicom is not recognizing ttyUSB0 (which was initially the means when I was using UART, as shown in the tutorial) and basically does not allow me to do it. Is there any other way?

Thank you.

tianfang-fafafa, 2022-09-16:
@vongracia If you can access to board via usb-keyboard, you will be able to obtain IP of board. So why dose not use ssh/scp to logon/transfer files from laptop to board ?

vongracia, 2022-09-18:
@tianfang-fafafa It is solved now. Thanks for your help.

Minicom is not needed indeed. I'm accesing the board by ssh/scp via ethernet cable.

closed 2022-09-26 by qianglin-xlnx