Xilinx/Vitis-AI #1019: Cannot import vitis_ai_library

JINYU-FAN, 2022-09-20:
I am trying to import several modules including vart, xir, and vitis_ai_library in python script, while vart and xir are imported successfully, I could not import vitis_ai_library. I have searched through the issues and found #737 which gives the download link of mpsoc:
https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc/VART/2021.2
Nevertheless, I am using U200 and U250, and I could not find similar download link under the folder https://github.com/Xilinx/Vitis-AI/tree/master/setup/alveo.

I also refer to Vitis AI Library User Guide (UG1354), where I found the link
https://www.xilinx.com/bin/public/openDownload?filename=vitis-ai-runtime-2.5.0.tar.gz
under Installation-Setting Up the Target-Step 3: Installing AI Library Package. I am not sure if it is the right link for vitis_ai_library module.

I tried to install the module following the instructions of Step 3: Installing AI Library Package, but failed because the files are rpm packages while my system is under ubuntu amd64. I used alien to generate deb version yet still not successful with installing the packages.

Here are my questions:
1. Am I referring to the correct link for importing vitis-ai-library in python?
2. If so, could you provide the deb packages?

Thank you in advance.

qianglin-xlnx, 2022-09-21:
HI @JINYU-FAN 
The following link is the package for edge platform not for cloud platform.
https://www.xilinx.com/bin/public/openDownload?filename=vitis-ai-runtime-2.5.0.tar.gz
For U200 and U250, maybe you can refer to: 
https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Library/README.md#running-vitis-ai-library-examples-on-alveo-u200alveo-u250

JINYU-FAN, 2022-09-21:
> HI @JINYU-FAN The following link is the package for edge platform not for cloud platform. https://www.xilinx.com/bin/public/openDownload?filename=vitis-ai-runtime-2.5.0.tar.gz For U200 and U250, [...]

Hi, thank you for replying!
I am trying to install the vitis-ai-library according to the link you offered:
https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Library/README.md#running-vitis-ai-library-examples-on-alveo-u200alveo-u250
However, I got stuck on the command:
./cmake.sh --clean --type=release --cmake-options=-DCMAKE_PREFIX_PATH=$CONDA_PREFIX --cmake-options=-DENABLE_DPUCADF8H_RUNNER=ON
which gives the error message:
[image]
[image]
I would really appreciate it if you could help me with this problem, because I have little experience with cmake.
Thank you.

JINYU-FAN, 2022-09-21:
> Hi, thank you for replying! I am trying to install the vitis-ai-library according to the link you offered: [...]

I have successfully run the Vitis-AI-Library examples in https://github.com/Xilinx/Vitis-AI/tree/master/src/Vitis-AI-Library#quick-start-for-cloud. I found that the command:
./cmake.sh --clean --type=release --cmake-options=-DCMAKE_PREFIX_PATH=$CONDA_PREFIX --cmake-options=-DENABLE_DPUCADF8H_RUNNER=ON
only works when no conda environment is activated.
However, I still cannot import vitis_ai_library, neither inside vitis-ai-pytorch environment nor outside. I wonder if there is any instructions to solve that problem.

Thank you.