Xilinx/Vitis-AI #1022: Just get one output after use pytorch to convert yolov4-tiny model

HuiJu1218, 2022-09-22:
Hi,
I had use vitis-ai 2.5 pytorch container to convert yolov4-tiny model. I had convert to xmodel successfully. But when I inference model on KV260 using python, I get just one output. Can anyone have idea? [Here is my model](https://drive.google.com/drive/folders/1eE1oAnRI8ZULulv8X2c8P76SlgPFxR_P?usp=sharing)
[image]
[image]

tuan3ihust, 2022-11-27:
Hello, did your compiled yolov4-tiny model has multiple DPU subgraph? 
I train yolov4-tiny with darknet and convert to caffe with Vitis AI 1.4 and after compiled, DPU subgraph number is 4
[image]

huisunCompiler, 2023-01-12:
Hi @HuiJu1218 ,
Thanks for providing xmodels, which helps a lot. From python xir I find 2 outputs, could you please describe the problem more in detail. (which output is missing.) many thanks :)
```
>>> import xir
>>> xir.Graph.deserialize("compiled.xmodel")
<xir.Graph named 'YoloBody'>
>>> g = xir.Graph.deserialize("compiled.xmodel")
>>> g.get_tail_ops()
{<xir.Op named 'YoloBody__YoloBody_Sequential_yolo_headP4__Conv2d_1__4248_fix_'>, <xir.Op named 'YoloBody__YoloBody_Sequential_yolo_headP5__Conv2d_1__4167_fix_'>}
>>>
```

huisunCompiler, 2023-01-12:
@tuanmonkeyy 
strided_slice-fix is not supported in vitis-ai 2.5 and 4 strided_slice-fix are partitioned to CPU. I recommend you update to vitis-ai release 3.0, the 4 strided_slice are optimized and the DPU subgraph is 1.

closed 2023-01-17 by qianglin-xlnx