Xilinx/Vitis-AI #1023: `vai_c_tensorflow2` compilation error when the `ignore_layers` option is specified in the function call `vitis_quantize.VitisQuantizer.quantize_model`

williamliao28, 2022-09-22:
### Description of the issue:
When the `ignore_layers` option is specified in the function call `vitis_quantize.VitisQuantizer.quantize_model`, attempt to compile the output quantized_model with `vai_c_tensorflow2` will result in the following error

[image]

### Steps to reproduce the issue:

 1. Quantize a float model with the `ignore_layers` options in the function call `vitis_quantize.VitisQuantizer.quantize_model` and save the quantized model:
 ```
quantizer = vitis_quantize.VitisQuantizer(float_model)
quantized_model = quantizer.quantize_model(calib_dataset=quant_dataset, fold_conv_bn=False, 
    ignore_layers=ignore_layers)
quantized_model.save(quant_model)
```
 2. Compile the quantized model with the following command:
```
vai_c_tensorflow2 \
            --model           build/quant_model/q_simple_nlCNN.h5 \
            --arch            $ARCH \
            --output_dir      build/compiled_$TARGET \
            --net_name        simple_nlCNN_$TARGET \
            --options '{"input_shape": "4,28,28,1"}'
``` 

### Possible cause of the error:
Layers listed in `ignore_layers` are not wrapped by any wrappers during quantization. In contrast, layer not in `ignore_layers` will be wrapped by either `VitisQuantize` or `QuantizeWrapper`. Thus, when the python script `tensorflow_translator.py` tries to execute

```
ksize = inner_layer_config.get("kernel_size")
```
in line 598, which is the previous function call before the error, the value of `ksize` will be `None` since `inner_layer_config` is still the config of previous quantized activation layer and a activation layer has no `kernel_size`.

zhenzhen-AMD, 2023-06-14:
Hi @williamliao28 ,
Could you please provide the float.h5 and quantized.h5 files? These would be very helpful in resolving the issue.

zhenzhen-AMD, 2023-07-13:
Hi @williamliao28 ，
Currently, **vai_c_tensorflow2** does not support quantized models generated using the **ignore_layers** option.

zhenzhen-AMD, 2023-07-26:
Closing this for now. Please re-open if there are any other concerns. Thanks.

closed 2023-07-26 by renjiangyong