Xilinx/Vitis-AI #1025: About Custom OP performance

tkkcc, 2022-09-23:
I am deploying a pytorch model to VCK190, which contains optical flow warping operator. 

From guide https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Custom-OP-Workflow, Custom OP can only be written as CPU OP. 

Also given samples https://github.com/Xilinx/Vitis-AI/tree/master/src/Vitis-AI-Library/cpu_task/ops, are written in simple for loop style. Seems not easy for further acceleration.

My questions are
1. Is the CPU Device on VCK190, or on connected PC.
2. How is the performance of custom op in sample code, compared with supported op. For example 10:1 runtime gap.
3. The guide said `an accelerated (PL or AIE) function` implementation can be used in CPU OP. Is there any sample code.
4. Is there a builtin optical flow warping operator.

Thanks!

lishixlnx, 2022-09-23:
1. vck190
2. you can test it by your self use the macro of __TIC__(some), you can find sample in vitis-ai-library source code 
3. there is no such sample currently
4. no

qianglin-xlnx, 2022-10-10:
Hi @tkkcc 
Is this issue solved?

closed 2022-11-01 by qianglin-xlnx