Xilinx/Vitis-AI #1030: How to specify the shape and the data_type of the output tensor of custom layer?

kayky233, 2022-09-26:
I imported Conv1d in my model, and quantized it. But when I compile, error occurs. I understand this error, but how to specify the shape and the data_type of the output tensor of this layer?

W0926 02:28:52.653378  2392 tool_function.cpp:171] [UNILOG][WARNING] The operator named conv1d_4, type: Conv1D, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)
F0926 02:28:52.653432  2392 wrapper.cpp:111] [UNILOG][FATAL][PYXIR_INVALID_DATA_TYPE][] Unsupported data type!
*** Check failure stack trace: ***
Aborted (core dumped)

environment: Vitis-AI 2.5, tensorflow 2.

yjtlab, 2023-05-06:
same problem

zhenzhen-AMD, 2023-06-14:
Hi @kayky233 and @yjtlab,

Based on your problem description, it is highly likely that you did not add **_'add_shape_info=True'_** during the quantization stage. Here's a reference example:

`
from tensorflow_model_optimization.quantization.keras import vitis_quantize

quantizer = vitis_quantize.VitisQuantizer(model)
quantized_model = quantizer.quantize_model(
    calib_dataset=train_images[0:10],
    include_cle=True,
    cle_steps=10,
    **add_shape_info=True**,
    include_fast_ft=True
)
`

zhenzhen-AMD, 2023-06-26:
@kayky233 
Could you provide code and model files?

zhenzhen-AMD, 2023-07-26:
Closing this for now. Please re-open if there are any other concerns. Thanks.

closed 2023-07-26 by renjiangyong