Xilinx/Vitis-AI #1031: How could find dpu_task.cpp?

yangyazhou97, 2022-09-26:
Hi Xilinx Community,
In  https://github.com/Xilinx/Vitis-AI/tree/master/examples/Vitis-AI-Library/samples/dpu_task/fadnet, I read fadnet.cpp ,  there is a line code " #include <vitis/ai/dpu_task.cpp", and I  only could find dpu.hpp in /usr/include/vitis/ai in ZCU102 images , but I also want to found out where is dpu.cpp ? 
Thanks.

qianglin-xlnx, 2022-09-27:
Hi @yangyazhou97 
Find the dpu.cpp in https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Library/dpu_task/src/dpu_task.cpp

yangyazhou97, 2022-09-27:
Yes, I could find dpu.cpp in github, but I want to find out where is dpu.cpp in ZCU102 images. Thanks

qianglin-xlnx, 2022-09-27:
It has been compiled as a library in ZCU102 image. You can find the libvitis_ai_library-dpu_task.so under /usr/lib.
If you want to debug it, you have to compile the Vitis-AI-Library in debug mode then install it and debug your program.

yangyazhou97, 2022-09-27:
I see, Thanks a lot.
I also have a question besides. In https://github.com/Xilinx/Vitis-AI/tree/master/examples/VART, there is many models about classification and video segmentation, but I have a model to do depth estimation by single image, how could I save the result through "cv::imwrite" or "cv::imshow", I learn two image depth estimation in https://github.com/Xilinx/Vitis-AI/tree/master/examples/Vitis-AI-Library/samples/dpu_task/fadnet, but it call “#include <vitis/ai/dpu_task.hpp>”， how could I save image by myself? Thanks.

lishixlnx, 2022-09-28:
in line 594 in fadenet.cpp, after 
    resize(final_img, ret, cv::Size(ret_width, ret_height));
use 
   cv::imwrite("some.jpg", ret);

yangyazhou97, 2022-09-28:
Yes, I know that. .Now I have a depth estimation model, 
I learned https://github.com/Xilinx/VitisAI/blob/master/examples/VART/squeezenet_pytorch/src/main.cc,  I enter ./squeezenet_pytorch  ./depth_estimation model ,how could I save depth image?

lishixlnx, 2022-09-29:
please make your question clear. 
what do you mean by "save" depth image? 
I can't understand what do you want.

yangyazhou97, 2022-09-29:
OK, there are many examples in https://github.com/Xilinx/Vitis-AI/tree/master/examples/VART, but many of them are classificaition models, after learning  https://github.com/Xilinx/VitisAI/blob/master/examples/VART/squeezenet_pytorch/src/main.cc, now I have a custom depth estimation model, how could I ouput depth image?

lishixlnx, 2022-10-04:
you can do it in something like this:
  cv::Mat mat( height, width, CV_32FC1);
  for(int i=0; i<height; i++) {
    for(int j=0; j<width; j++) {
      mat.ptr<float>(i)[j] =  some_value_your_logic_calculated_from_model;
    }
  }
cv::imwrite("result_mat.png", mat);

yangyazhou97, 2022-10-04:
Thank you so much, I think that's what I'm doing, but I still wrong the "image3". I post my code below.
#include <assert.h>
#include <dirent.h>
#include <stdio.h>
#include <stdlib.h>
#include <sys/stat.h>
#include <unistd.h>

#include <cassert>
#include <cmath>
#include <cstdio>
#include <fstream>
#include <iomanip>
#include <iostream>
#include <queue>
#include <string>
#include <vector>
#include <time.h>
#include <chrono>
#include <algorithm>

#include "common.h"
/* header file OpenCV for image processing */
#include <opencv2/opencv.hpp>

using namespace std;
using namespace cv;
using std::chrono::high_resolution_clock;
using std::chrono::milliseconds;

GraphInfo shapes;

const string baseImagePath = "./images/";
const string wordsPath = "./";

/**
 * @brief put image names to a vector
 *
 * @param path - path of the image direcotry
 * @param images - the vector of image name
 *
 * @return none
 */
void ListImages(string const& path, vector<string>& images) {
  images.clear();
  struct dirent* entry;

  /*Check if path is a valid directory path. */
  struct stat s;
  lstat(path.c_str(), &s);
  if (!S_ISDIR(s.st_mode)) {
    fprintf(stderr, "Error: %s is not a valid directory!\n", path.c_str());
    exit(1);
  }

  DIR* dir = opendir(path.c_str());
  if (dir == nullptr) {
    fprintf(stderr, "Error: Open %s path failed.\n", path.c_str());
    exit(1);
  }

  while ((entry = readdir(dir)) != nullptr) {
    if (entry->d_type == DT_REG || entry->d_type == DT_UNKNOWN) {
      string name = entry->d_name;
      string ext = name.substr(name.find_last_of(".") + 1);
      if ((ext == "JPEG") || (ext == "jpeg") || (ext == "JPG") ||
          (ext == "jpg") || (ext == "PNG") || (ext == "png")) {
        images.push_back(name);
      }
    }
[... 140 lines cut ...]
  LOG(INFO) << "create running for subgraph: " << subgraph[0]->get_name();
  /*create runner*/
  auto runner = vart::Runner::create_runner(subgraph[0], "run");
  // ai::XdpuRunner* runner = new ai::XdpuRunner("./");
  /*get in/out tensor*/
  auto inputTensors = runner->get_input_tensors();
  auto outputTensors = runner->get_output_tensors();

  /*get in/out tensor shape*/
  int inputCnt = inputTensors.size();
  int outputCnt = outputTensors.size();
  TensorShape inshapes[inputCnt];
  TensorShape outshapes[outputCnt];
  shapes.inTensorList = inshapes;
  shapes.outTensorList = outshapes;
  getTensorShape(runner.get(), &shapes, inputCnt, outputCnt);

  int thread_number = 20;
  float total_frames = thread_number * 654;

  runSqueezenet(runner.get());

 
  return 0;
}

yangyazhou97, 2022-10-04:
Actually, I'm not sure if the "result" is real output.

lishixlnx, 2022-10-10:
I know what you mean.   The result is correct.
This image is  Mat(CV_32FC1),  not the common image of Mat(CV_8UC3).
The CV_32FC1 is used only by the opencv innerly use to store binary data,
not like the common picture that we can see.
the pictures that we can see normally is CV_8UC3 ( or CV_8UC1)。

you can change your CV_32FC1 to CV_8UC1 in order to show by imshow().
that need some special technical skill( for example, you need map the big float data to 0-255 int8 number)

yangyazhou97, 2022-10-10:
Thank you so much, I have solved this question. I will try using the method you said. Thanks again.

closed 2022-10-10 by yangyazhou97