Xilinx/Vitis-AI #1033: Bus error when running code in the board

githubcyy, 2022-09-28:
I followed the steps to quantify, compile, and deploy on my  board, but a Bus error appeared when running the code in the board. I'm targeting the wrong location, it's "vart.Runner.create_runner(subgraph, "run")". I don't know what's wrong, is there something wrong with the compiled xmodel? Could anybody give me some advice? Thanks!

qianglin-xlnx, 2022-09-28:
Hi @githubcyy 
Could you try to run an VART example on your board, such as https://github.com/Xilinx/Vitis-AI/tree/master/examples/VART/resnet50?
Bus error usually caused by the application program. You can also try to use 'xdputil' to run your custom model. If you can run your custom model successfully on your custom board, then "bus error" issue is caused by your application program not the model.

githubcyy, 2022-09-30:
Thank you @qianglin-xlnx 
I tried to run an VART example on my board according to VART/resnet50, but "bus error" also appeared. I used ACU4EV, Vitis AI1.3.
[image]

qianglin-xlnx, 2022-10-10:
Hi @githubcyy 
In VAI1.3, there is no 'xdputil' tool. Could you try to update the VAI1.3 to the latest version?

githubcyy, 2022-10-12:
Thank you @qianglin-xlnx 
I have some doubts about 'xdputil' tool,  Is this tool integrated in the latest version? Does this tool need to be installed separately?  I found the deploy code import xir and vart, how does this  'xdputil' tool work?

qianglin-xlnx, 2022-10-12:
Hi @githubcyy 
Yes. 'xdputil' tool is integrated in the VAI1.4 and later. I suggest to upgrade to the latest version VAI2.5.
The tool is integrated in Vitis-AI-Library. When you install the Vitis-AI-LIbrary, the xdputil will be installed by default.
Xdputil tool can be used to check the dpu. For more about xdputil, refer to https://docs.xilinx.com/r/2.0-English/ug1354-xilinx-ai-sdk/Using-the-xdputil-Tool

qianglin-xlnx, 2022-11-14:
Closing since no activity for more than 1 month, please reopen if you still have question, thanks

closed 2022-11-14 by qianglin-xlnx