Xilinx/Vitis-AI #1035: Vitis AI 2.5 Board overload issue

akshayvignesh, 2022-09-28:
Hi, 
I tried to execute the Vitis AI 2.5 on U50 lv board. I was able to infer the results for the sample images but when I try to execute the Performance tests my System reboots. What may be the reason for the unexpected reboot? Any leads would be appreciated. 

Command used:

./test_performance_classification resnet50 test_performance_classification.list -t 8 -s 60

System Info:

OS: Ubuntu 18.04.1
kernel: 4.15.0-193-generic
Card :U50LV

SuZhan0322, 2022-10-12:
Hi @akshayvignesh ,
The System reboots is due to power consumption. And due to power consumption, the alveo card of U50/U50lv needs to be downclocked to 60% to run the model multi-threading test.  Please refer to the command to reduce the frequency: https://docs.xilinx.com/r/2.0-English/ug1354-xilinx-ai-sdk/Scaling-Down-the-Frequency-of-the-DPU
Thanks!

SuZhan0322, 2022-11-02:
Great! @akshayvignesh I am glad to hear that. And Please reopen the issue if you have any other questions! thanks!

closed 2022-11-02 by SuZhan0322