Xilinx/Vitis-AI #1036: Error when making inference on ZCU102 - Vitis 1.4

vongracia, 2022-09-28:
Hi folks,

I've followed this https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.4/Design_Tutorials/07-yolov4-tutorial (using **./docker_run.sh xilinx/vitis-ai-cpu:1.4.916** because it requires Vitis 1.4) to quantize and compile a darknet trained yolov4 on my custom data. I tried the caffe tutorial. When i get to the final step to make the inference on the FPGA ZCU102 I get this error:

**[libprotobuf ERROR google/protobuf/text_format.cc:323] Error parsing text-format vitis.ai.proto.DpuModelParamList: 1:5: Message type "vitis.ai.proto.DpuModelParamList" has no field named "name".
WARNING: Logging before InitGoogleLogging() is written to STDERR
F1119 17:30:48.142455  1033 configurable_dpu_task_imp.cpp:142] Check failed: ok cannot parse config file. config_file=dpu_yolov4_voc_andante.prototxt
*** Check failure stack trace: ***
Aborted**

Do you know what this mean? Please give me a hand, it driving me crazy.

Here the 4 files (file.prototxt, file.xmodel, md5sum.txt, meta.json) produced at the end. I used only the prototxt and xmodel to run the inference: https://wetransfer.com/downloads/032432a82712d0d99a67cc0c2d6beabb20220928144618/267f8e

Please take a look, maybe you can find something
Thanks

lishixlnx, 2022-09-30:
you use the wrong .prototxt file in last step.
you should use this one :
[dpu_yolov4.prototxt](https://github.com/Xilinx/Vitis-AI-Tutorials/blob/1.4/Design_Tutorials/07-yolov4-tutorial/dpu_yolov4/dpu_yolov4.prototxt)

vongracia, 2022-09-30:
@lishixlnx thanks

The file you sent is an already defined file, isnt it?.  The thing is that I trained my own yolov4 darknet model on my custom data and I quantized/compiled it for inference on ZCU102. This is why I got the specific .prototxt I sent you.

If I need to use the new .prototxt you provided, how shall I modify it to make it work along with my my .xmodel to make inference?

vongracia, 2022-09-30:
If I do what you says @lishixlnx and put my .xmodel and .prototxt in my ZCU102 and run inference:

./test_jpeg_yolov4 dpu_yolov4_voc_andante.xmodel test_image.jpg 

The output is:
root@xilinx-zcu102-20221:~/examples/yolov4# ./test_jpeg_yolov4 dpu_yolov4_voc_andante.xmodel test_image.jpg 
**WARNING: Logging before InitGoogleLogging() is written to STDERR
W1119 17:34:36.245939  1087 dpu_runner_base_imp.cpp:676] CHECK fingerprint fail ! model_fingerprint 0x1000020f6014407 dpu_fingerprint 0x101000016010407 
F1119 17:34:36.246059  1087 dpu_runner_base_imp.cpp:648] fingerprint check failure.
*** Check failure stack trace: ***
Aborted**

Do you know how to solve this? @lishixlnx  @qianglin-xlnx

Thanks!

qianglin-xlnx, 2022-10-01:
Hi @vongracia 
For the fingerprint check failure error, you need you recompile the model with the corresponding dpu.arch file.

vongracia, 2022-10-03:
Please give me a hand @qianglin-xlnx 

I followed this tutorial: https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.4/Design_Tutorials/07-yolov4-tutorial 
And converted from darknet to caffe.

The thing is that quantization process generates a .prototxt that is totally different from the file @qianglin-xlnx  provided above.
Tutorial generates this format: layer {
  name: "data"
  type: "Input"
  top: "data"
#  transform_param {
#    mirror: false
#    yolo_height: 416
#    yolo_width: 416
#  }
  input_param {
    shape {
      dim: 1
      dim: 3
      dim: 416
      dim: 416
    }
  }
}
layer {
  name: "data_fixed"
  type: "FixedNeuron"
  bottom: "data"
  top: "data"
  param {
    lr_mult: 0
    decay_mult: 0
  }
  phase: TRAIN
  fixed_param {
    fixed_method: OVER_FLOW
    bit_width: 8
    follow_data_layer: true
  }
}
And the file by @lishixlnx this format:
model {
  name: "yolov4"
  kernel {
     name: "yolov4"
     mean: 0.0
     mean: 0.0
     mean: 0.0
     scale: 0.00390625
     scale: 0.00390625
     scale: 0.00390625
  }
  model_type : YOLOv3
  yolo_v3_param {
    num_classes: 1
    anchorCnt: 3
    layer_name: "109"
    layer_name: "101"
    layer_name: "93"
    conf_threshold: 0.25
    nms_threshold: 0.45
...

What is the right one? Why is this happening? Please give me a hand. I don't manage to make my darknet model make inference on a ZCU102.

THANKS

qianglin-xlnx, 2022-10-08:
Hi @vongracia 
According to https://github.com/Xilinx/Vitis-AI/issues/1052, it seems that you have solved this issue and run the inference successfully?

vongracia, 2022-10-08:
Yes, it is solved. Thanks

closed 2022-10-14 by qianglin-xlnx