Xilinx/Vitis-AI #1041: docker build gpu fail on V1.4.1

zhihao8841, 2022-10-01:
I trying build gpu Vitis ai 1.4.1

error no matching distribution found for tensorflow2.3, under the figure 
[image]

The python3.9 not supply tensorflow 2.3. if i want do build vitis ai 1.4.1 how can i do?
does anyone have any suggestion?

denishem, 2022-10-12:
Hi, I think the problem is that Tensorflow 2.3.0 is compatible up to Python 3.8 as you can see in the table [here](https://www.tensorflow.org/install/source#gpu) , while the GPU Dockerfile installs the latest version of Miniconda3, which is based on Python 3.9.
This could probably be fixed by changing this line:
https://github.com/Xilinx/Vitis-AI/blob/c6d79ee83ee940e83ececc9fac693fd9bf1a515c/setup/docker/docker/DockerfileGPU#L244
With
```
    && wget --progress=dot:mega https://repo.anaconda.com/miniconda/Miniconda3-py38_4.12.0-Linux-x86_64.sh -O miniconda.sh \
```

I am facing the same issue, the image is still building so I cannot be sure it fixes the problem.

zhihao8841, 2022-10-13:
Thank you, I success build Vitis ai docker .Vitis-ai-gpu:1.4.1.978
[image]

gaoxinghua951, 2023-05-24:
> 

请问您就是按照上面的方法解决的吗？

zhihao8841, 2023-05-24:
> > 
> 
> 请问您就是按照上面的方法解决的吗？

是的 參考上面建議的方法

gaoxinghua951, 2023-05-24:
> > > 
> > 
> > 
> > 请问您就是按照上面的方法解决的吗？
> 
> 是的 參考上面建議的方法

我在构建pytoch版本gpu2.5时遇见上述问题，使用上述方法后还是不行，