Xilinx/Vitis-AI #1045: DPU IP debug performance counters

MaistoV, 2022-10-04:
I am currently working with the DPUCZDX8G IP, v3.4 and v4.0. I noticed that there are 9 registers per DPU core not documented in PG338 (up to
latest version v4.0).
These are perfomance counters and have debug purposed for the IP engineers as discussed in a mail exchange with the source maintainer, Ye Yang. Who also pointed me here.

All the documentation I could find is in the source code of the header file of dpu.ko kernel module, available in the linux-xlnx GitHub repo
(https://github.com/Xilinx/linux-xlnx/blob/master/drivers/misc/xlnx_dpu.h). In particular, this file contains:
- the offset mapping macros: DPU_P_END_C, DPU_C_­END_C, ..., DPU_CYCLE_H;
- a brief description of "struct ioc_kernel_run_t", for the fields: counter, cend_cnt, send_cnt, ..., lstart_cnt.

Is it possible find some more detailed documenta­tion of these regist­ers?
In particular, I am interested in:
- primarly, semant­ics of "misc", "conv­", "load" and "save" registers, namely the operati­ons they count;
- difference betwe­en "start" and "end" values;
- consistency prop­erties (e.g., when are these registers updated w.r.t the corresponding operations).

Thanks for the support.

qianglin-xlnx, 2023-02-20:
Hi @MaistoV 
So far as I know, there is no document described these registers.

qianglin-xlnx, 2023-03-27:
Closing since no activity for more than 1 month, please reopen if you still have any questions, thanks

closed 2023-03-27 by qianglin-xlnx