Xilinx/Vitis-AI #1048: How to save quantized model in resnet18_qat.py?

yangyazhou97, 2022-10-05:
In https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_qat.py, I want to save quantized model, but it can't be loaded correctly, I notice there is "def save_checkpoint".

niuxjxlnx, 2022-10-10:
@yangyazhou97 :

In deployable.pth, model parameters are still float, but this float in different from that of model before quantization. Here the float parameters are (int*quantization scale).

yangyazhou97, 2022-10-10:
Sorry, I don't get your point, what do you mean by int*quantization scale? I want to know the model parameter after quantization.

yuwang-xlnx, 2022-10-19:
Hi @yangyazhou97,

The quantization process is done in forward pass and the parameters after quantization are not stored in model.state_dict().

yangyazhou97, 2022-10-19:
Hi, @yuwang-xlnx ,
If I want to see the quantized parameters, how should I do?

yuwang-xlnx, 2022-10-19:
> Hi, @yuwang-xlnx , If I want to see the quantized parameters, how should I do?

There is no particularly straightforward way to do this. You can refer to the code: 
https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/quantization/modules/conv_fused.py#L307

yangyazhou97, 2022-10-19:
I added the code "print("w_quantized:{}".format(w_quantized))" in conv_fused.py, then I python resnet18_qat.py, I don't see the quantized output.

yuwang-xlnx, 2023-01-13:
> I added the code "print("w_quantized:{}".format(w_quantized))" in conv_fused.py, then I python resnet18_qat.py, I don't see the quantized output.

Where did you add the print statement? Is your model in training mode or evaluation mode?

yangyazhou97, 2023-02-12:
@yuwang-xlnx  I add the print statement in here, https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/quantization/modules/conv_fused.py#L307

yuwang-xlnx, 2023-02-13:
Hi @yangyazhou97,

The statement wil print the quantized weights when your model is in training mode.

yangyazhou97, 2023-02-13:
Hi @yuwang-xlnx, 
Can't I output the quantized weights if I use Post Training Quantization? how about Quantization Aware Training?

yuwang-xlnx, 2023-02-13:
> Hi @yuwang-xlnx, Can't I output the quantized weights if I use Post Training Quantization? how about Quantization Aware Training?

Yes, the statement only works in Quantization Aware Training, not in Post Training Quantization.

yangyazhou97, 2023-02-13:
I tried to output the quantization Quantization Aware Training before, but it seems that it still didn't work, the output model was still float numbers.

yuwang-xlnx, 2023-02-13:
> I tried to output the quantization Quantization Aware Training before, but it seems that it still didn't work, the output model was still float numbers.

The fake quantization is used in Vitis AI quantizer, that means the tensors are not rellay integer numbers and reprensented by floating numbers.

yangyazhou97, 2023-02-13:
So can't I print the saved model parameters in integer numbers and how could I testify I have really finished the quantization?

yuwang-xlnx, 2023-02-14:
> So can't I print the saved model parameters in integer numbers and how could I testify I have really finished the quantization?

Please refer to the user guide: https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Running-Quantization-and-Getting-the-Result

closed 2023-03-01 by niuxjxlnx