Xilinx/Vitis-AI #1049: Missing Implementation of `tanh` in `__create_model_from_tf2`

cabreraam, 2022-10-05:
I'm trying to take a sequential model through the Vitis AI quantizer and compiler. One of the layers in my model uses `tanh` as an activation function.

In the implementation for creating an xmodel from a TensorFlow2 model (located in `/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py` in the container that I'm using), there is missing support for the `tanh` activation function in the `__create_model_from_tf2` member function of class `TFTranslator`.

In the `if/elif/else` statements that check for `op_types`, there is support for other activation functions like `relu` 
```python
elif op_type in ["relu", "ReLU"]:
...
    xnode = XModelNodeRelu(wrapper_name)
...
````

and `sigmoid`:
```python
elif op_type in ["sigmoid", "Sigmoid"]:
...
    xnode = XModelNodeSigmoid(wrapper_name)
...
```

but for some reason, `tanh` is not there.  Can someone add the implementation for this?

Thanks!

konegen, 2022-12-13:
Hey @cabreraam I have the same problem.
Do you know if there is a solution or a workaround right now.
I am also searching how I can implement a TF2 model with 'tanh' activation function?

Thanks!

cabreraam, 2022-12-13:
Hi @konegen 

I found out that, unfortunately, [`tanh` is not supported in the Tensorflow flow of Vitis-AI 2.5](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Operators-Supported-by-TensorFlow). I was directed to try the [Custom OP workflow](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Custom-OP-Workflow), but I never tried doing that. Instead, I changed my activation function just to get something running (which is __not ideal__ but it's what I did to make forward progress). Hopefully this is helpful!

zhenzhen-AMD, 2023-06-14:
Hi @cabreraam and @konegen ,
Currently the compiler does not support tanh. Even if tanh is supported through the Custom Op Workflow, ultimately tanh will still be deployed to the CPU.

zhenzhen-AMD, 2023-06-26:
Hi @cabreraam and @konegen ,
This issue is open for a long time, we will close it , if you still have some problem about this issue, you can reopen it, Thank you.

closed 2023-06-26 by qianglin-xlnx