Xilinx/Vitis-AI #1050: How to enable `logger` at `DEBUG` level

cabreraam, 2022-10-05:
I'm not familiar with Python, but I see that VItis-AI uses the native Python `logger` library statements. I see statements like the following in the codebase:

```python
logger.debug(f"create XModelNodeSigmoid object: name: {xnode.op_name}")
```

The above statement is in `/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py`, so this is part of the Vitis-AI compiler utility.

If I'm using the command-line utility `vai_c_tensorflow2`, how can I trigger the `DEBUG` level of logging and where can I find the logs?

zhenzhen-AMD, 2023-06-14:
Hi @cabreraam ,
Modify line 51 in the file "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/xconverter.py" from "enable = False" to "enable = True". After doing this, you will be able to get the log records in the '~/.xnnc_logs/*.log' files.

zhenzhen-AMD, 2023-06-26:
Hi @cabreraam ,
This issue is open for a long time, we will close it , if you still have some problem about this issue, you can reopen it, Thank you.

closed 2023-06-26 by qianglin-xlnx