Xilinx/Vitis-AI #1053: export  XLNX_ENABLE_DEVICES does not work

jzymessi, 2022-10-09:
Hi xilinx:
     
     I used two vck5000 to run the program, according to the flow 

     export XLNX_ENABLE_DEVICES=0,1 
     
     When I run the official test_performance_yolovx program in Vitis-AI-Library, it was useful, but when I ran the program I wrote, I found that it did not run on two cards. 

     May I ask how to solve the problem?

SuZhan0322, 2022-10-14:
Hi @jzymessi ,
There is a processing code for "XLNX_ENABLE_DEVICES" in the xilinx vart to select the device to be called. 
The first thing need to confirm is that your code is calling the api of vart? If yes, please check if xrm is installed correctly on the host. please refer to the link: https://github.com/Xilinx/Vitis-AI/blob/master/setup/alveo/scripts/install_xrm.sh

SuZhan0322, 2023-06-13:
Please reopen this issue if still have questions. Thanks!