Xilinx/Vitis-AI #1055: Quantization with pytorch yolov5

alpereenkaratas, 2022-10-09:
I want to quantize the Pytorch yolov5 model. But I don't know how to fill in quantize.py and other necessary files. I couldn't find any tutorial on this topic. The transactions I mentioned in the #675 issue have been done successfully. I want to do exactly what was done there. Can anyone help me on what to do about this?

wangxd-xlnx, 2022-10-13:
Hi @alpereenkaratas 

Thanks for your interest in Vitis-AI toolchain. Due to the license issue, we have not publicly released yolov5 related content for the time being.
But in the next version Vitis-AI 3.0, we plan to give more reference information for latest yolo series including v5 and v6. The expected release time is in 2023.1.

qianglin-xlnx, 2022-11-02:
Closing since no activity for more than 2 weeks, please reopen if you still have question, thanks

closed 2022-11-02 by qianglin-xlnx

IkrameBeggar, 2023-04-25:
> I want to quantize the Pytorch yolov5 model. But I don't know how to fill in quantize.py and other necessary files. I couldn't find any tutorial on this topic. The transactions I mentioned in the [...]

Hello @alpereenkaratas I am facing the same problem and I don´t know how to fill the quant.py file. Did you find any resources regarding that?