Xilinx/Vitis-AI #1056: [UNILOG][FATAL][XIR_REMOVE_OP_FAIL][Failed to remove an op!] Cannot remove xir::Op, HopeNet

marenan, 2022-10-10:
Hi, I use vitis ai 2.5 compile hopenet, mistake as:
"""
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
[UNILOG][INFO] Graph name: HopeNetLite, with op num: 754
[UNILOG][INFO] Begin to compile...
[0;31m[UNILOG][FATAL][XIR_REMOVE_OP_FAIL][Failed to remove an op!] Cannot remove xir::Op{name = HopeNet__HopeNet_Sequential_globalpool__AvgPool2d_0__15866_sink_transpose_16_fix, type = fix} from graph. Because there is at least one op who takes its output as input.
[m*** Check failure stack trace: ***
Aborted (core dumped)
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
"""

someone can help me?

huisunCompiler, 2022-11-03:
Hello @marenan ,

we have fixed this bug in our internal develop version. You now can modify the model using the python code as below. The code copies the transpose and reshapes three times which fed as the input of 3 matmuls. The inserted transpose and reshapes will be removed in the compiling procedure. It is verified this modified model will not decrease the model performance. 

The red circle indicates the position of this bug. 
[image]

put this code in the folder storing "Hopenet_int.xodel" and this code will give a modified model "modify_Hopenet_int.xmodel".

Use the modify_Hopenet_int.xmodel to complete the remaining procedures. 

import xir
g = xir.Graph.deserialize("Hopenet_int.xmodel")
ops = g.toposort()
def copy_op(g, op, name_append):
    return g.create_op(op.get_name() + name_append, op.get_type(), op.get_attrs(), op.get_input_ops())
for op in ops:
    if op.get_name() == "Hopenet__Hopenet_8293_sink_transpose_0":
        transpose = op, copy_op(g, op, "_0"), copy_op(g, op, "_1")
    if op.get_name() == "Hopenet__Hopenet_input_shape_attr_1_":
        reshape_attr1 = op, copy_op(g, op, "_0"), copy_op(g, op, "_1")
    if op.get_name() == "Hopenet__Hopenet_input_shape_attr_0_":
        reshape_attr0 = op, copy_op(g, op, "_0"), copy_op(g, op, "_1")
    if op.get_name() == "Hopenet__Hopenet_input_shape_i0":
        reshape_shape = op, copy_op(g, op, "_0"), copy_op(g, op, "_1")
        reshape_shape[1].replace_input_ops(reshape_attr0[0], reshape_attr0[1])
        reshape_shape[2].replace_input_ops(reshape_attr0[0], reshape_attr0[2])
        reshape_shape[1].replace_input_ops(reshape_attr1[0], reshape_attr1[1])
        reshape_shape[2].replace_input_ops(reshape_attr1[0], reshape_attr1[2])
    if op.get_name() == "Hopenet__Hopenet_input":
        reshape = op, copy_op(g, op, "_0"), copy_op(g, op, "_1")
        reshape[1].replace_input_ops(reshape_shape[0], reshape_shape[1])
        reshape[2].replace_input_ops(reshape_shape[0], reshape_shape[2])
        reshape[1].replace_input_ops(transpose[0], transpose[1])
        reshape[2].replace_input_ops(transpose[0], transpose[2])
    if op.get_name() == "Hopenet__Hopenet_Linear_fc_pitch__8302":
        op.replace_input_ops(reshape[0], reshape[1])
    if op.get_name() == "Hopenet__Hopenet_Linear_fc_roll__8303":
        op.replace_input_ops(reshape[0], reshape[2])
g.serialize("modify_Hopenet_int.xmodel")

qianglin-xlnx, 2023-01-17:
Closing since no activity for more than 2 months, please reopen if you still have any questions, thanks

closed 2023-01-17 by qianglin-xlnx