Xilinx/Vitis-AI #1057: PyTorch YOLOv5 quantizer error

alpereenkaratas, 2022-10-10:
I am trying to quantize yolov5 with vitis ai 2.5. While the line of code seen in the image in the quantizer.py script is running, I am getting the error in the image below. Any ideas on the solution?

[image]
[image]

carrotandjzy, 2022-10-12:
Can you move the post process that contains the "meshgrid" Ops out of the forward process?

carrotandjzy, 2022-10-12:
This multi-outputs op can't be deployed.

alpereenkaratas, 2022-10-12:
[common.txt](https://github.com/Xilinx/Vitis-AI/files/9762195/common.txt)
[quantize.txt](https://github.com/Xilinx/Vitis-AI/files/9762196/quantize.txt)

@carrotandjzy  I don't fully understand what you are talking about. I shared the codes. Can you describe the arrangement you are talking about over the code? What exactly do I need to do?

Note: The `test(quantized_model, device, test_loader)` part in the first photo I shared will actually be `evaluate(quantized_model, device, test_loader)`. Sorry for posting the wrong photo.

niuxjxlnx, 2022-10-18:
@alpereenkaratas :

Please refer to the comments in https://github.com/Xilinx/Vitis-AI/issues/674.

useruser2023, 2024-02-19:
@alpereenkaratas  have you solved this problem. Care to share the solution.