Xilinx/Vitis-AI #1058: Compile Error：The fix info is error or unsupported

Windhelm-H, 2022-10-11:
Hello 
    I encountered a problem when compiling with Vitis-AI 2.5. The log as

    [UNILOG][INFO] Compile mode: dpu
    [UNILOG][INFO] Debug mode: function
    [UNILOG][INFO] Target architecture: DPUCVDX8G_ISA3_C32B6
    [UNILOG][INFO] Graph name: IFNet, with op num: 320
    [UNILOG][INFO] Begin to compile...
    [0;31m[UNILOG][FATAL][XCOM_UNSUPPORT_QUANTIZATION][The fix info is error or unsupported.] elew0's quantization info's variation is too large.
    [m*** Check failure stack trace: ***
    Aborted (core dumped)

    how can I fix it？ many think

niuxjxlnx, 2023-01-13:
@Windhelm-H ,

Can you upload your result of quantization?

closed 2023-03-02 by niuxjxlnx

CNStanLee, 2025-06-13:
Hi, I met this again. The result of the quantization is attached below.
Thanks!!

[quant_info.json](https://github.com/user-attachments/files/20727438/quant_info.json)