Xilinx/Vitis-AI #1060: VAI trace Vitis AI analyzer results issues

micro83, 2022-10-13:
I did a fine_grained tracing for a small model compiled from PyTorch and I noticed the followings: 

1- the data of the first layer is not correct, for example the workload , LdWB belongs to the whole model itself not to the first layer. 
moreover,I'm not able to discover where  the numbers of LdFM & StFM came from. Therefore, the Performance & AvgBw are incorrectly calculated.
I discovered this mismatch from comparing the numbers with cnn_dump_xmodel.txt and from manual calculations.
Could you please suggest a quick fix for this mismatch?

2- may I know the formulas used to calculate the workload for Conv2d, MaxPool2d, and Linear layers? I tried the ones used in CNN literature but they didn't match.

3- I suggest to make it possible changing the units from the configuration files, because the rounded numbers in the report are loosing the accuracy drastically in the case of the small models. for example : the real Workload of a layer is 415904 operations which rounded in the report to ~0 GOP !
I have made some modifications in the vart_parser.py to change the unites and present the numbers without rounding.

4- When importing the files in Vitis Analyzer :Its not clear to which data types belong DDR read & write diagrams exactly, DDRC_PORT_S1 to S5?
I mean which one belongs to DPU instruction, feature map and weight+biase data?

please find here the vi trace results, and the attached cnn_dump_xmodel.txt for comparison.
[image]

[cnn_dump_xmodel.txt](https://github.com/Xilinx/Vitis-AI/files/9775283/cnn_dump_xmodel.txt)

li900309, 2022-10-21:
Q4: 
[image]
find the detail in page 380 of [ug1085](https://docs.xilinx.com/r/en-US/ug1085-zynq-ultrascale-trm)

It's hard to discern that information, from the architecture diagram we can see that, S1 - S5 represent of five ingress ports of ddr controller, that is in the last level of data path, data traffic can be mixed in many places of the data path

I think there is one possible ways to make the data more accurate
vaitrace get axi traffic data from harden AXI performance monitor in DDR controller, In order to get more accurate information, we can add AXI Performance Monitor IP in DPU data ports, but this consumes additional PL resources

li900309, 2022-10-21:
Q3: great idea and do you have a pull request for this change ?

qianglin-xlnx, 2023-06-26:
Closing since no activity for more than half year, please reopen if you still have question, thanks.

closed 2023-06-26 by qianglin-xlnx