Xilinx/Vitis-AI #1063: Operator usage method for shape, slice and expand...

Shreyas-NR, 2022-10-18:
Hi,
I'm trying to deploy my quantized model on **DPUCZDX8G_ISA1_B4096** target. 

I needed help with the usage of the operators. 
1. I am unsure of what operator to use for computing the below operations on the CPU before passing the tensor to the DPU.

I'm using the model quantized with pytorch. 
My application code is using python wrapper. 

`self.module_0 = py_nndct.nn.Input()` **Input Tensor dimension: {1,1,288,288}**
` self.module_1 = py_nndct.nn.Module('shape')` **Input Tensor dimension: {1,1,288,288}**
`  self.module_2 = py_nndct.nn.Module('shape')` **Input Tensor dimension: {1,1,288,288}**
  `self.module_3 = py_nndct.nn.Module('shape')` **Input Tensor dimension: {1,1,288,288}**
  `self.module_4 = py_nndct.nn.strided_slice()` **Don't know what this operation does**
  `self.module_5 = py_nndct.nn.expand()` **Don't know what this operation does** however output tensor shape is {1,3,288,288}
  `self.module_6 = py_nndct.nn.Conv2d(in_channels=3, out_channels=64, kernel_size=[7, 7], stride=[2, 2], padding=[3, 3], dilation=[1, 1], groups=1, bias=True)` Input Tensor dimension: {1,288,288, 3} 
  
[image]
  
2. Should I implement the fix2float, float2fix, transpose, and reshape-fix operator functionality within my application script? If so, could you please advise how to implement them? Else if built-in operators are present, how can I make use of the operator's APIs in my python application script?

Thanks,

lishixlnx, 2022-10-18:
2  fix2float, float2fix, transpose, and reshape-fix are already implemented.  
    You needn't do anything for it.  It is used automatically by graph runner. 
    Graph runner found this is a cpu op and try to find the corresponding lib. 
   for example,  one of the lib is in /usr/lib/libvart_op_imp_float2fix.so.
3. you may ignore the warning now. 
   for the error message,  A2J_model__A2J_model_ResNetBackBone_Backbone__input_1 is not recognized. If you insist to use it, you must implement it as customer-op.
  you can refer to https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Custom-OP-Registration?tocId=eFDclUqkGVyQE6AwM_Um2Q  for detail about customer op

Shreyas-NR, 2022-10-18:
Hi @lishixlnx 
Thank you for your quick response.

I am not clear whether to use graph runner or vart.runner.

My application code looks like this now, 
what tensor should I pass to the DPU input tensor?
My input image dimensions are {1,1,288,288}
input to the DPU tensor is {1,288,288,3}

Any help is appreciable.

```
from ctypes import *
from typing import List
import cv2
import numpy as np
import vart
import vitis_ai_library
import os
import pathlib
import xir
import threading
import time
import sys
import argparse
import json

_divider = '-------------------------------'
cropWidth = 288
cropHeight = 288
Img_mean = 3
Img_std = 2

with open('./labels.json') as f:
    dict = json.load(f)
    annotations_test = dict

keypointsWorldtest = []
keypointsPixeltest = []
bndbox_test = []
for key,value in dict.items():
    if key != 'intrinsics':
        temp = value[0]
        keypointsWorldtest.append(temp['3d_joints'])
        keypointsPixeltest.append(temp['2d_joints'])
        bndbox_test.append(temp['bbox'])

keypointsWorldtest = np.asarray(keypointsWorldtest, dtype=np.float32)
keypointsPixeltest = np.asarray(keypointsPixeltest, dtype=np.float32)
bndbox = np.asarray(bndbox_test, dtype=np.float32)
TestImgFrames = len(keypointsPixeltest)

test_image_ids = list(annotations_test.keys())
imageOutputs = np.ones((cropHeight, cropWidth, 1), dtype="float32")
def preprocess_fn(image_path, fix_scale, index):
    '''
    Image pre-processing.
    Opens image as grayscale, adds channel dimension, normalizes to range 0:1
    and then scales by input quantization scaling factor
    input arg: path of image file
    return: numpy array
    '''
    img_int = np.load(image_path).astype(np.float32)
    new_Xmin = max(bndbox[index][0] , 0)
    new_Ymin = max(bndbox[index][1] , 0)
    new_Xmax = min(bndbox[index][2] , img_int.shape[1] - 1)
    new_Ymax = min(bndbox[index][3] , img_int.shape[0] - 1)

    imCrop_int = img_int.copy()[int(new_Ymin):int(new_Ymax), int(new_Xmin):int(new_Xmax)] 
[... 267 lines cut ...]

# only used if script is run as 'main' from command line
def main():

    # construct the argument parser and parse the arguments
    ap = argparse.ArgumentParser()  
    ap.add_argument('-d', '--image_dir', type=str, default='images', help='Path to folder of images. Default is images')  
    ap.add_argument('-t', '--threads',   type=int, default=1,        help='Number of threads. Default is 1')
    ap.add_argument('-m', '--model',     type=str, default='A2J_kv260.xmodel', help='Path of xmodel. Default is CNN_kv260.xmodel')
    args = ap.parse_args()  

    print ('Command line options:')
    print (' --image_dir : ', args.image_dir)
    print (' --threads   : ', args.threads)
    print (' --model     : ', args.model)

    app(args.image_dir,args.threads,args.model)

if __name__ == '__main__':
  main()

```

I have attached the subgraph below
[image]

closed 2022-10-25 by Shreyas-NR

AdrFebles, 2023-01-19:
Hi!
I have a xmodel with DPU operations followed by some transpose and fix2floats, you can see a part of the scheme bellow:
[image]

I used "graph_runner" mode to execute all the operations compiled in the .xmodel including both DPU and CPU operations. I expected to see an increase in the frames per second ratio compared to what was obtained with the "runner" mode, which only runs DPU operations so the CPU ones needed to be implemented in the app.
Following are the performance results in each case:

**Graph_runner** mode: (FPS=0.61, total frames = 100.00 , time=164.941214 seconds)

**Runner** mode: (FPS=5.32, total frames = 100.00 , time=18.801671 seconds)

@Shreyas-NR i don't know if you tried with both modes?
Can anyone help me understand how graph_runner works and why it takes so long to inference my model?

Thanks!

hocken-li, 2023-07-07:
Hi, @Shreyas-NR 
I meet the same problem with you in expand.
Can I know how to solve this? How to  set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)??
` type: expand, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)`

Shreyas-NR, 2023-12-21:
Hi @hocken-li
I could figure out the solution to this warning message. Did you solve it?