Xilinx/Vitis-AI #1068: Missing dpu.xclbin in pre-build SD Image

eerobert, 2022-10-24:
Would like to know where can I find the dpu.xclbin in the pre-build SD image for KV260?

Thanks

eerobert, 2022-10-27:
Hi, anyone can help?

Without the dpu.xclbin for KV260, I cannot go thru the Vitis-AI VART example.

When I run the Vitis-AI Resnet50 example, it complains the following...

//
root@xilinx-kv260-starterkit-20221:~/Vitis-AI/examples/VART/resnet50# ./resnet50 /usr/share/vitis_ai_library/models/resnet50/resnet50.xmodel 
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1026 19:24:33.753268  1682 main.cc:292] create running for subgraph: subgraph_conv1
F1026 19:24:33.851476  1682 xrt_bin_stream.cpp:60] Check failed: fd_ > 0 (-1 vs. 0) , open(/run/media/mmcblk0p1/dpu.xclbin) failed.
*** Check failure stack trace: ***
Aborted

//

Thanks

eerobert, 2022-10-27:
Hi,

I think I found the solution to the problem

1. If you run thru the optional steps in readme, it will update the /etc/vart.conf and point the dpu.xclbin to /run/media/mmcblk0p1. Can modify the file to point to original xclbin at /lib/firmware/xilinx/kv260-benchmark-b4096/kv260-benchmark-b4096.xclbin

2. If you skip all the optional steps in readme, VART example can run.

closed 2022-10-27 by eerobert

mj1028, 2024-12-24:
Hi,
I think I got the same issue as u ever encounter then.
The Log as following,
xilinx-kv260-starterkit-20222:/usr/share/vitis_ai_library/samples/classification$ sudo ./test_jpeg_classification ~/resnet50_pt/resnet50_pt.xmodel  ~/images/001.jpg 
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1223 18:10:41.553745  2170 xrt_bin_stream.cpp:47] Please check your /etc/vart.conf
Its format should be :
    firmware: xx
Example:
    firmware: /run/media/mmcblk0p1/dpu.xclbin
F1223 18:10:41.553920  2170 xrt_bin_stream.cpp:51] [UNILOG][FATAL][VART_OPEN_DEVICE_FAIL][Cannot open device] open(/run/media/mmcblk0p1/dpu.xclbin) failed.
*** Check failure stack trace: ***
Aborted

I've go through using Vivado process to create project so I did not with xclbin file.
About the second u mentioned "skip all the optional steps in readme, VART example can run"---> what this mean ? could u help me? thx a lot then!!