Xilinx/Vitis-AI #1070: How to use the operators in CPU mode

Shreyas-NR, 2022-10-25:
Hi,

I have a graph as shown below

[image]

**How to use these CPU operators on the DPU output tensor in CPU mode?**

I'm using `vart.Runner.create_runner(dpu_subgraph, "run")`

Thank you

lishixlnx, 2022-11-01:
you can implement these op in customer op mode.
please refer to 
https://docs.xilinx.com/r/en-US/ug1354-xilinx-ai-sdk/Custom-Operator-Implementation-and-Registration

Shreyas-NR, 2022-11-01:
Hi @lishixlnx,

thanks for the reply,

I believe that these operators are prebuilt. 

[image]

I needed to know how to use them in my python application code to run on board.

lishixlnx, 2022-12-07:
you need use the graph-runner mode. 
Please refer to below link:
https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Deployment?tocId=hOzwBJoHpy5vmUxppUrc6Q

qianglin-xlnx, 2023-01-12:
Closing since no activity for more than 1 month, please reopen if you still have any questions, thanks

closed 2023-01-12 by qianglin-xlnx