Xilinx/Vitis-AI #1071: Quantinized with  VITIS AI for yolov3 model generated quant_info with all null

cpracticer, 2022-10-26:
### This is my quantize code
def quantize(build_dir,quant_mode,batchsize):

  dset_dir = build_dir + '/dataset'
  float_model = build_dir + '/float_model'
  quant_model = build_dir + '/quant_model'

  # use GPU if available   
  if (torch.cuda.device_count() > 0):
    print('You have',torch.cuda.device_count(),'CUDA devices available')
    for i in range(torch.cuda.device_count()):
      print(' Device',str(i),': ',torch.cuda.get_device_name(i))
    print('Selecting device 0..')
    device = torch.device('cuda:0')
  else:
    print('No CUDA devices available..selecting CPU')
    device = torch.device('cpu')

  # load trained model
  model = YoloBody([[6, 7, 8], [3, 4, 5], [0, 1, 2]],20,pretrained = False).to(torch.device('cpu'))
  pre=torch.load('epoch_weights.pth',map_location=torch.device('cpu'))
  model.load_state_dict(pre)
  #device = torch.device("cuda:0" if torch.cuda.is_available() else "cpu")
  #weigths = torch.load('epoch_weights.pth',map_location='cpu')
  #model = weigths['model']
  #model = model.half().to(device)

  # force to merge BN with CONV for better quantization accuracy
  optimize = 1

  # override batchsize if in test mode
  if (quant_mode=='test'):
    batchsize = 1
  
  rand_in = torch.randn([batchsize, 3, 416, 416])
  quantizer = torch_quantizer(quant_mode, model, (rand_in), output_dir=quant_model) 
  quantized_model = quantizer.quant_model

  # data loader
  train_set = val_dataset
  #train_loader = torch.utils.data.DataLoader(train_set, batch_size=1, shuffle=False)  # , num_workers=8
  #test_loader = torch.utils.data.DataLoader(train_set, batch_size=batchsize, shuffle=False,collate_fn=yolo_dataset_collate)
  test_loader = torch.utils.data.DataLoader(train_set, shuffle = False, batch_size = batchsize, num_workers = 8, pin_memory=True,drop_last=True, collate_fn=yolo_dataset_collate, sampler=None)
                                              # evaluate 
  #test(quantized_model, device, test_loader)
  

  # export config
  if quant_mode == 'calib':
    quantizer.export_quant_config()
  if quant_mode == 'test':
    quantizer.export_xmodel(deploy_check=False, output_dir=quant_model)
  
  return
  #

### This is my log file
PyTorch version :  1.10.1
3.7.12 | packaged by conda-forge | (default, Oct 26 2021, 06:08:53)
[GCC 9.4.0]
-----------------------------------------
 Command line options:
--build_dir    :  ./build
--quant_mode   :  calib
--batchsize    :  100
-----------------------------------------
No CUDA devices available..selecting CPU

[VAIQ_WARN]: CUDA is not available, change device to CPU

[... 507 lines cut ...]
[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.1.conv.bias

[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.2.conv.weight

[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.2.conv.bias

[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.3.conv.weight

[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.3.conv.bias

[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.4.conv.weight

[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.4.conv.bias

[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.5.conv.weight

[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.5.conv.bias

[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.6.weight

[VAIQ_WARN]: Parameter tensor is not quantized: YoloBody::last_layer2.6.bias

[VAIQ_WARN]: Quantization is not performed completely, check if model inference function is called!!!

### How to slove it? Thanks for all

cpracticer, 2022-10-26:
this code was modified by code of https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.4/Design_Tutorials/09-mnist_pyt
it's calib quant_mode

niuxjxlnx, 2022-11-14:
@cpracticer ,

The reason to report this message is the quantizer.quant_model was not inferenced with your dataloader.
You should inference quantizer.quant_model with code:

cpracticer, 2022-11-14:
Thanks!As you said，I need to do like use
_model=quantizer
output=model(testloader),_
It will quantize my model.
I have solved it!
Thanks again

closed 2022-11-14 by niuxjxlnx

gaoxinghua951, 2023-06-07:
Hello, how did you solve it? I had the same problem as you when quantifying yolox

cpracticer, 2023-06-07:
hello,dear gaoxinghua951,the main reason for this problem is that you didn't do 'test'.Please refer Vitis-AI-tutorials-->09 minit_pyt-->common.py--> test 

def test(model, device, test_loader):
    '''
    test the model
    '''
    model.eval()
    test_loss = 0
    correct = 0
    with torch.no_grad():
        for data, target in test_loader:
            data, target = data.to(device), target.to(device)
            output = model(data)
            pred = output.argmax(dim=1, keepdim=True)
            correct += pred.eq(target.view_as(pred)).sum().item()

    acc = 100. * correct / len(test_loader.dataset)
    print('\nTest set: Accuracy: {}/{} ({:.2f}%)\n'.format(correct, len(test_loader.dataset), acc))

    return

there is a step "output=model(data)" is critical.
Best

cpracticer, 2023-06-07:
The steps of Vitis AI Quantization is similar,but different Vitis AI version is incompatible, you can have a try.I didn't use Vitis AI 2.5.
Best

gaoxinghua951, 2023-06-07:
> Vitis AI 量化的步骤类似，但不同的 Vitis AI 版本不兼容，可以试一试。我没有使用Vitis AI 2.5。 最好

Thank you very much for your reply. I think I have found a solution according to your prompt!