Xilinx/Vitis-AI #1072: Run resnet50.xmodel to display CHECK fingerprint fail! model_fingerprint 0x1000020f7014402 dpu_fingerprint 0x101000016010402

yxb0054410, 2022-10-27:
On the target board, I checked the information about my dpu with the xdputil query command.(My target board model is the XCZU4EV series)
xdputil query
{
    "DPU IP Spec":{
        "DPU Core Count":1,
        "IP version":"v4.0.0",
        "enable softmax":"False"
    },
    "VAI Version":{
        "libvart-runner.so":"Xilinx vart-runner Version: 2.5.0-c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8  2022-10-20-14:23:29 ",
        "libvitis_ai_library-dpu_task.so":"Xilinx vitis_ai_library dpu_task Version: 2.5.0-c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8  2022-06-15 07:33:00 [UTC] ",
        "libxir.so":"Xilinx xir Version: xir-c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8 2022-10-20-13:18:02",
        "target_factory":"target-factory.2.5.0 c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8"
    },
    "kernels":[
        {
            "DPU Arch":"DPUCZDX8G_ISA1_B1024",
            "DPU Frequency (MHz)":325,
            "cu_idx":0,
            "fingerprint":"0x101000016010402",
            "is_vivado_flow":true,
            "name":"DPU Core 0"
        }
    ]
}
Matches my arch.json information

 download from making the git clone - recurse vitis ai - submodules https://github.com/Xilinx/Vitis-AI

And follow the tutorial. Because the target board is custom (XCZU4EV). So at the time of compilation model, I need to change/opt/vitis_ai/compiler/arch/DPUCZDX8G ZCU104 / arch. Json.

And I changed the arch.json file to

{

"Target" : "DPUCZDX8G_ISA1_B1024"

}

Then compile the resnet50 model. There is an error
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B1024
[UNILOG][FATAL][TARGET_FACTORY_UNREGISTERED_TARGET][Unregistered target!] Cannot find target with name DPUCZDX8G_ISA1_B1024,
 valid names are: {DPUCADF8H_ISA0=>0x700000000000000,DPUCAHX8H_ISA2=>0x20200000000002a,
DPUCAHX8H_ISA2_ELP2=>0x20200000000002e,DPUCAHX8L_ISA0=>0x30000000000001d,
DPUCAHX8L_ISA0_SP=>0x30000000000101d,DPUCVDX8G_ISA1_C16B1_QT=>0x601001432022110,
DPUCVDX8G_ISA1_C32B3=>0x601001036088131,DPUCVDX8H_ISA1_F1W2=>0x501000000007ff6,
DPUCVDX8H_ISA1_F2W2=>0x501000000000fee,DPUCVDX8H_ISA1_F2W2_6PE=>0x501000000000c2e,
DPUCVDX8H_ISA1_F2W2_6PE_NONCASCADE=>0x50100000000082e,DPUCVDX8H_ISA1_F2W4_6PE=>0x501000000020c2e,
DPUCVDX8H_ISA1_F2W4_6PE_DWC=>0x501000000020c2f,DPUCVDX8H_ISA1_F3W0=>0x501000000029c36,
DPUCVDX8H_ISA1_F3W1=>0x501000000031c36,DPUCVDX8H_ISA1_F4W2=>0x501000000002c36,
DPUCVDX8H_ISA1_F8W2=>0x501000000006c36,DPUCZDI4G_ISA0_B4096_DEMO_SSD=>0x400002003220206,
DPUCZDI4G_ISA0_B8192D8_DEMO_SSD=>0x400002003220207,DPUCZDX8G_ISA0_B1024_MAX=>0x1000020f7014402,
DPUCZDX8G_ISA0_B1024_MIN=>0x100002022010102,DPUCZDX8G_ISA0_B1152_MAX=>0x1000020f7012203,
DPUCZDX8G_ISA0_B1152_MIN=>0x100002022010103,DPUCZDX8G_ISA0_B1600_MAX=>0x1000020f7014404,
DPUCZDX8G_ISA0_B1600_MIN=>0x100002022010104,DPUCZDX8G_ISA0_B2304_MAX=>0x1000020f7014405,
DPUCZDX8G_ISA0_B2304_MAX_BG2=>0x1000020f6014405,DPUCZDX8G_ISA0_B2304_MIN=>0x100002022010105,
DPUCZDX8G_ISA0_B3136_MAX=>0x1000020f7014406,DPUCZDX8G_ISA0_B3136_MAX_BG2=>0x1000020f6014406,
DPUCZDX8G_ISA0_B3136_MIN=>0x100002022010106,DPUCZDX8G_ISA0_B4096_MAX=>0x1000020f7014407,
DPUCZDX8G_ISA0_B4096_MAX_BG2=>0x1000020f6014407,DPUCZDX8G_ISA0_B4096_MAX_EM=>0x1000030f7014407,
DPUCZDX8G_ISA0_B4096_MIN=>0x100002022010107,DPUCZDX8G_ISA0_B512_MAX=>0x1000020f7012200,
DPUCZDX8G_ISA0_B512_MIN=>0x100002022010100,DPUCZDX8G_ISA0_B800_MAX=>0x1000020f7012201,
DPUCZDX8G_ISA0_B800_MIN=>0x100002022010101
}
*** Check failure stack trace: ***
This program has crashed!
Aborted (core dumped)

Please tell me how to solve the above problem. (Since the chip of my target board is XCZU4EV), I can't directly adopt arch.json of ZCU104. Change it to the corresponding arch.json. According to the error message, I guess that my arch.json is not in complier

qianglin-xlnx, 2022-10-27:
Please update the arch.json with the following content.  And compile the model with the new arch.json.
{
    "fingerprint": "0x101000016010402"
}

yxb0054410, 2022-10-27:
Thank you very much for your reply.

I update the arch.json file as follows: 

{ 

"fingerprint": "0x101000016010402" 

} 

Then continue to compile the model, the compilation process has a new problem: 

UNILOG][INFO] Compile mode: dpu 

[UNILOG][INFO] Debug mode: function 

[UNILOG][FATAL][XCOM_INVALID_TARGET][Invalid DPU target] Compiler backend cannot support DPUCZDX8G_ISA1. (valid list:  DPUCADF8H_ISA0 DPUCAHX8H_ISA2 DPUCAHX8L_ISA0 DPUCVDX8G_ISA1 DPUCVDX8H_ISA1 DPUCZDI4G_ISA0 DPUCZDX8G_ISA0) 

*** Check failure stack trace: *** 

This program has crashed! 

Aborted (core dumped)

Shreyas-NR, 2022-10-27:
What is your VITIS version on the HOST PC?

make sure you have version 2.5 docker built.

yxb0054410, 2022-10-28:
Thank you very much for your reply. 

My VITIS version is 2022.1. 

docker pull VITIS AI's latest version

qianglin-xlnx, 2022-11-01:
Hi @yxb0054410 
It doesn't look like you are using the VAI2.5 docker. "DPUCZDX8G_ISA1" is only supported in VAI2.5. 
When you launch the docker, you will find the docker version, as shown below. Could you please check it again? 
[image]

qianglin-xlnx, 2022-11-16:
Hi @yxb0054410 
Is this issue solved?

godfatehr, 2022-11-21:
I have exactly the same problem. 
I am using Vitis AI 2.0 and whole project is based on it. At the moment I cannot switch to the higher version of Vitis AI. How can I solve the issue?

qianglin-xlnx, 2022-11-21:
Hi @godfatehr  
Could you post your error log?
If you use Vitis AI 2.0, then just use the Vitis AI 2.0 docker to compile your model.

godfatehr, 2022-11-21:
[image]

godfatehr, 2022-11-21:
I have a custom board and tried to use the arch.json file which generated from DPU_TRD to compile the ResNet18.xmodel after successful quantization process

qianglin-xlnx, 2022-11-21:
Hi @godfatehr 
For Vitis AI2.0, you can use Vitis AI 2.0 docker not Vitis AI 2.5 docker. Then the compiling issue will be solved.

godfatehr, 2022-11-21:
Indeed, thanks for your support.

closed 2023-01-12 by qianglin-xlnx