Xilinx/Vitis-AI #1073: Error to load the Xmodel on the board

Afef00, 2022-10-27:
I ran Vitis AI CPU docker container from host using ./docker_run.sh xilinx/vitis-ai-cpu:1.3.411. After successfully training my custom model using the following code

```
import os
import numpy as np
import tensorflow as tf
from sklearn.metrics import confusion_matrix
image_ordering_dim="tf"
def preprocess(X_train, X_test):
    """
    Convert from float64 to float32 and normalize normalize to decibels
    relative to full scale (dBFS) for the 4 sec clip.
    """
    X_train = X_train.astype('float32')
    X_test = X_test.astype('float32')

    X_train = np.array([(X - X.min()) / (X.max() - X.min()) for X in X_train])
    X_test = np.array([(X - X.min()) / (X.max() - X.min()) for X in X_test])
    return X_train, X_test
def prep_train_test(X_train, y_train, X_test, y_test, nb_classes):
    """
    Prep samples ands labels for Keras input by noramalzing and converting
    labels to a categorical representation.
    """
    print('Train on {} samples, validate on {}'.format(X_train.shape[0],
                                                       X_test.shape[0]))

    # normalize to dBfS
    X_train, X_test = preprocess(X_train, X_test)

    # Convert class vectors to binary class matrices
    Y_train = tf.keras.utils.to_categorical(y_train, nb_classes)
    Y_test = tf.keras.utils.to_categorical(y_test, nb_classes)

    return X_train, X_test, Y_train, Y_test
def keras_img_prep(X_train, X_test, img_dep, img_rows, img_cols):

    X_train = X_train.reshape(X_train.shape[0], img_rows, img_cols, 1)
    X_test = X_test.reshape(X_test.shape[0], img_rows, img_cols, 1)
    input_shape = (img_rows, img_cols, 1)

    return X_train, X_test, input_shape

def cnn(X_train, y_train, X_test, y_test, batch_size,
        nb_classes, epochs, input_shape):
    """
  
    """

    inputs = tf.keras.Input(shape=input_shape)
    x = tf.keras.layers.Conv2D(128, (3,3), activation='relu', input_shape=(input_shape),padding='same')(inputs)
    x = tf.keras.layers.MaxPooling2D((4,3),strides=(1, 3))(x)
    x = tf.keras.layers.Conv2D(32, (1,3), activation='relu',input_shape=(input_shape),padding='same')(x)
    x = tf.keras.layers.MaxPooling2D((1,3), strides=(1, 3))(x)
    x = tf.keras.layers.Flatten()(x)
    x = tf.keras.layers.Dense(512, activation='relu')(x)
    x = tf.keras.layers.Dense(512, activation='relu')(x)
    x = tf.keras.layers.Dropout(0.2)(x)
    outputs = tf.keras.layers.Dense(2, activation='softmax')(x)

    model = tf.keras.Model(inputs=inputs, outputs=outputs, name='classifier')
    
    model.compile(loss='categorical_crossentropy',
                  optimizer='adadelta',
                  metrics=['accuracy'])

    history = model.fit(X_train, y_train, batch_size=batch_size, epochs=epochs,
                        verbose=1, validation_data=(X_test, y_test))

    # Evaluate accuracy on test and train sets
[... 105 lines cut ...]
opener = urllib.request.build_opener()
opener.addheaders = [('User-agent', 'Mozilla/5.0')]
urllib.request.install_opener(opener)
from pynq_dpu import DpuOverlay
overlay = DpuOverlay("dpu.bit")
overlay.load_model("depression_test.xmodel")
```

---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-2-4783d6fd0d7f> in <module>
      1 from pynq_dpu import DpuOverlay
      2 overlay = DpuOverlay("dpu.bit")
----> 3 overlay.load_model("depression_test.xmodel")
      4 # overlay.load_model("depression_flatten_model.xmodel")

/usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq_dpu/dpu.py in load_model(self, model)
    169             self.graph = xir.Graph.deserialize(abs_model)
    170             subgraphs = get_child_subgraph_dpu(self.graph)
--> 171             assert len(subgraphs) == 1
    172             self.runner = vart.Runner.create_runner(subgraphs[0], "run")

AssertionError: 

I don't get what's the problem

qianglin-xlnx, 2022-10-29:
Hi @Afef00 
Do you use PYNQ system image?
If not, you can refer to https://github.com/Xilinx/Vitis-AI/blob/1.3.1/demo/VART/inception_v1_mt_py/inception_v1.py python example.

Afef00, 2022-10-29:
@qianglin-xlnx Yes, I used PYNQ-DPU (PYNQ image 2.7). The problem is when I tried other models, such as those provided on Jupyter notebook or other Xmodels that I tested before, it works fine. I don't get what the problem is.

qianglin-xlnx, 2022-10-29:
From the log you provide, the model you compiled has 2 DPU subgraphs.
`[UNILOG][INFO] Total device subgraph number 5, DPU subgraph number 2`

While your program assert at the number of subgraph, which the DPU subgraph must be 1 in your program.  Maybe you can modify this in your code.

--> 171 assert len(subgraphs) == 1

Afef00, 2022-10-29:
@qianglin-xlnx How can I modify this in the code?

Afef00, 2022-10-31:
I tried to modify the model architecture as follow:

  ```
  inputs = tf.keras.Input(shape=input_shape)
    x = tf.keras.layers.Conv2D(128, (3,3), activation='relu', input_shape=(input_shape))(inputs)
    x = tf.keras.layers.MaxPooling2D((4,3))(x)
    x = tf.keras.layers.Conv2D(32, (1,3), activation='relu')(x)
    x = tf.keras.layers.MaxPooling2D((1,3))(x)
    x = tf.keras.layers.Flatten()(x)
    x = tf.keras.layers.Dense(512, activation='relu')(x)
    x = tf.keras.layers.Dense(512, activation='relu')(x)
    outputs = tf.keras.layers.Dense(2, activation='softmax')(x)

    model = tf.keras.Model(inputs=inputs, outputs=outputs, name='classifier')
```

However, when I tried to compile the model I got this error:

**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['quantized_depression_V2/quantize_eval_model.pb'], model_type='tensorflow', out_filename='./depression_V2_org.xmodel', proto=None)
[INFO] tensorflow model: quantized_depression_V2/quantize_eval_model.pb
[INFO] parse raw model     :100%|█| 26/26 [00:00<00:00, 27134.09it/s]           
[INFO] infer shape (NHWC)  :100%|█| 33/33 [00:00<00:00, 5077.85it/s]            
[INFO] infer shape (NHWC)  :100%|█| 30/30 [00:00<00:00, 4038.81it/s]            
[INFO] generate xmodel     :100%|█| 30/30 [00:01<00:00, 17.32it/s]              
[INFO] generate xmodel: /workspace/depression_V2_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20221031-153814-245"
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Graph name: quantize_eval_model, with op num: 44
[UNILOG][INFO] Begin to compile...
[UNILOG][WARNING] xir::Op{name = max_pooling2d_1/MaxPool, type = pool-fix} has been allocated to CPU.
[UNILOG][Check Failed: bank_addr_end >= min_runtime_size][/home/xbuild/conda-bld/xcompiler_1606385139459/work/src/pass/passes/GenInst/AssignBankGroup.imp:1858][XCOM_ALLOCATE_BANK_FAIL][There is not enough bank space for the tensor] 
*** Check failure stack trace: ***
Aborted (core dumped)

qianglin-xlnx, 2022-11-16:
Hi @Afef00 
Sorry about the late reply. Is this issue solved?
My team does not have PYNQ-DPU. If you still have problem, you can ask PYNQ-DPU team for help.