Xilinx/Vitis-AI #1076: Quantizer cannot handle VGG16 Model in Pytorch

alewarne, 2022-10-28:
Hi,

I am trying to quantize the VGG16 model in pytorch using the following code:

```
import torch
import torchvision
from pytorch_nndct.apis import torch_quantizer

def quantize_model():
    orig_model = torchvision.models.vgg16()
    rand_in = torch.randn((1, channels, rows, cols))
    quantizer = torch_quantizer('calib', orig_model, (rand_in))
    quantized_model = quantizer.quant_model
    print(quantized_model)
    print(orig_model)
    quantized_model.eval()
    y = quantized_model(rand_in)
    quantizer.export_quant_config()

if __name__ == '__main__':
    quantize_model()
```

The resulting output is given by

```
[VAIQ_NOTE]: Loading NNDCT kernels...

[VAIQ_WARN]: CUDA is not available, change device to CPU

[VAIQ_NOTE]: Quant config file is empty, use default quant configuration

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cpu'.

[VAIQ_NOTE]: =>Parsing VGG...

[VAIQ_NOTE]: Start to trace model...

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 42/42 [00:00<00:00, 4473.36it/s, OpInfo: name = return_0, type = Return]                                           

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quant_output/VGG.py)

[VAIQ_NOTE]: =>Get module with quantization.
VGG(
  (module_0): deephi_Input()
  (module_1): deephi_Conv2d(3, 64, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_2): deephi_ReLU(inplace=True)
  (module_3): deephi_Conv2d(64, 64, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_4): deephi_ReLU(inplace=True)
  (module_5): deephi_MaxPool2d(kernel_size=[2, 2], stride=[2, 2], padding=[0, 0], dilation=[1, 1], ceil_mode=False)
  (module_6): deephi_Conv2d(64, 128, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_7): deephi_ReLU(inplace=True)
  (module_8): deephi_Conv2d(128, 128, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_9): deephi_ReLU(inplace=True)
  (module_10): deephi_MaxPool2d(kernel_size=[2, 2], stride=[2, 2], padding=[0, 0], dilation=[1, 1], ceil_mode=False)
  (module_11): deephi_Conv2d(128, 256, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_12): deephi_ReLU(inplace=True)
  (module_13): deephi_Conv2d(256, 256, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_14): deephi_ReLU(inplace=True)
  (module_15): deephi_Conv2d(256, 256, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_16): deephi_ReLU(inplace=True)
  (module_17): deephi_MaxPool2d(kernel_size=[2, 2], stride=[2, 2], padding=[0, 0], dilation=[1, 1], ceil_mode=False)
  (module_18): deephi_Conv2d(256, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_19): deephi_ReLU(inplace=True)
  (module_20): deephi_Conv2d(512, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_21): deephi_ReLU(inplace=True)
  (module_22): deephi_Conv2d(512, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_23): deephi_ReLU(inplace=True)
  (module_24): deephi_MaxPool2d(kernel_size=[2, 2], stride=[2, 2], padding=[0, 0], dilation=[1, 1], ceil_mode=False)
  (module_25): deephi_Conv2d(512, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_26): deephi_ReLU(inplace=True)
  (module_27): deephi_Conv2d(512, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_28): deephi_ReLU(inplace=True)
  (module_29): deephi_Conv2d(512, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
  (module_30): deephi_ReLU(inplace=True)
  (module_31): deephi_MaxPool2d(kernel_size=[2, 2], stride=[2, 2], padding=[0, 0], dilation=[1, 1], ceil_mode=False)
  (module_32): deephi_AvgPool2d(kernel_size=[6, 6], stride=[0, 0], padding=[0, 0])
  (module_33): DeephiFuncModule('flatten')
  (module_34): deephi_Linear(in_features=25088, out_features=4096, bias=True)
  (module_35): deephi_ReLU(inplace=True)
  (module_36): deephi_Linear(in_features=4096, out_features=4096, bias=True)
  (module_37): deephi_ReLU(inplace=True)
  (module_38): deephi_Linear(in_features=4096, out_features=1000, bias=True)
)
VGG(
  (features): Sequential(
    (0): Conv2d(3, 64, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (1): ReLU(inplace=True)
    (2): Conv2d(64, 64, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (3): ReLU(inplace=True)
    (4): MaxPool2d(kernel_size=2, stride=2, padding=0, dilation=1, ceil_mode=False)
    (5): Conv2d(64, 128, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (6): ReLU(inplace=True)
    (7): Conv2d(128, 128, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (8): ReLU(inplace=True)
    (9): MaxPool2d(kernel_size=2, stride=2, padding=0, dilation=1, ceil_mode=False)
    (10): Conv2d(128, 256, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (11): ReLU(inplace=True)
    (12): Conv2d(256, 256, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (13): ReLU(inplace=True)
    (14): Conv2d(256, 256, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (15): ReLU(inplace=True)
    (16): MaxPool2d(kernel_size=2, stride=2, padding=0, dilation=1, ceil_mode=False)
    (17): Conv2d(256, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (18): ReLU(inplace=True)
    (19): Conv2d(512, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (20): ReLU(inplace=True)
    (21): Conv2d(512, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (22): ReLU(inplace=True)
    (23): MaxPool2d(kernel_size=2, stride=2, padding=0, dilation=1, ceil_mode=False)
    (24): Conv2d(512, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (25): ReLU(inplace=True)
    (26): Conv2d(512, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (27): ReLU(inplace=True)
    (28): Conv2d(512, 512, kernel_size=(3, 3), stride=(1, 1), padding=(1, 1))
    (29): ReLU(inplace=True)
    (30): MaxPool2d(kernel_size=2, stride=2, padding=0, dilation=1, ceil_mode=False)
  )
  (avgpool): AdaptiveAvgPool2d(output_size=(7, 7))
  (classifier): Sequential(
    (0): Linear(in_features=25088, out_features=4096, bias=True)
    (1): ReLU(inplace=True)
    (2): Dropout(p=0.5, inplace=False)
    (3): Linear(in_features=4096, out_features=4096, bias=True)
    (4): ReLU(inplace=True)
    (5): Dropout(p=0.5, inplace=False)
    (6): Linear(in_features=4096, out_features=1000, bias=True)
  )
)
Traceback (most recent call last):
  File "quantization_pytorch.py", line 38, in <module>
    cli()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "quantization_pytorch.py", line 30, in quantize_model
    y = quantized_model(rand_in)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "quant_output/VGG.py", line 81, in forward
    output_module_0 = self.module_32(output_module_0)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/nn/modules/avgpool.py", line 41, in forward
    output = super().forward(qinput)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/torch/nn/modules/pooling.py", line 617, in forward
    self.padding, self.ceil_mode, self.count_include_pad, self.divisor_override)
RuntimeError: stride should not be zero
```
It seems that `(module_32): deephi_AvgPool2d(kernel_size=[6, 6], stride=[0, 0], padding=[0, 0])` in the quantized model is the problem since it is an AvgPool Layer with stride 0 although it should be an AdaptiveAvgPool2d layer. Does someone know why this conversion error is happening and how to solve it efficiently?

niuxjxlnx, 2022-11-14:
@alewarne,

I cannot repeat your issue in Vitis-AI 2.5 and channels=3, rows=224, cols=224. What are your Vitis-AI version and values of channels, rows and cols?

niuxjxlnx, 2022-11-17:
In Vitis-AI 3.0, protection for corner rows<224, cols=224 will be added in Vgg16.

closed 2023-01-13 by niuxjxlnx