Xilinx/Vitis-AI #1077: vai_q_tensorflow quantize : unsupported input shape dimension

Afef00, 2022-10-28:
Hello,
I ran Vitis AI CPU docker container from host using ./docker_run.sh xilinx/vitis-ai-cpu:1.3.411. After successfully training my custom model with the following summary
[image]
When I tried to quantize the model as follow:
```

!vai_q_tensorflow quantize \
    --input_frozen_graph frozen_graph_epilepsy.pb \
    --input_fn input_func.calib_input \
    --output_dir quantized_epilepsy \
    --input_nodes input_4 \
    --output_nodes activation_16/Softmax \
    --input_shapes ?,1,21,55,114\
    --calib_iter 32
```

I got the following error:

Traceback (most recent call last):

 File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_q_tensorflow", line 11, in <module>

  sys.exit(run_main())

 File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 1030, in run_main

  app.run(main=my_main, argv=[sys.argv[0]] + unparsed)

 File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/platform/[app.py](https://app.py/)", line 40, in run

  _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)

 File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/[app.py](https://app.py/)", line 303, in run

  _run_main(main, args)

 File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/[app.py](https://app.py/)", line 251, in _run_main

  sys.exit(main(argv))

 File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 1029, in <lambda>

  my_main = lambda unused_args: main(unused_args, FLAGS)

 File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 651, in main

  input_shapes = _parse_input_shapes(input_nodes, flags.input_shapes)

 File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 230, in _parse_input_shapes

  'Input_shapes should be 4-dimension int shape list (support unknown batch_size), e.g. ?,224,224,3 or 1,299,299,3'

ValueError: Input_shapes should be 4-dimension int shape list (support unknown batch_size), e.g. ?,224,224,3 or 1,299,299,3

 

Any ideas how to solve this please?

Regards

foolmarks, 2022-10-28:
--input_shapes ?,1,21,55,114\    ...5 dimensions?

Afef00, 2022-10-28:
Yes I am trying to implement this work on ZCU104 board

zhenzhen-AMD, 2023-06-27:
Hi @Afef00 ，
Currently, the quantization tool **vai_q_tensorflow** only supports models with 4-dimensional inputs. However, support for models with 5-dimensional inputs will be added in the future.

zhenzhen-AMD, 2023-07-10:
Hi @Afef00 ,
I will close this issue. If you have any new questions in the future, please feel free to open a new issue. Thank you.

closed 2023-07-10 by renjiangyong