Xilinx/Vitis-AI #1078: Pytorch yolov5 quantization error

shoayi, 2022-10-29:
Hi!iI have some questions to ask you.This error occurs when i run this commond"python resnet18_quant.py --quant_model float" to quantize yolov5 model(use the code of "ultralytics"):
[image]
Have you encountered the same situation?The weights I trained and the code I modified are available at the following link:
链接：https://pan.baidu.com/s/1plB4lEBGayeDwmCjGAIW5Q 
提取码：mnox 
--来自百度网盘超级会员V5的分享

niuxjxlnx, 2022-11-14:
@shoayi :

It seems that the network shape inference has problem? how about the float model inference? does it work or not?

niuxjxlnx, 2023-01-13:
@shoayi ,

In your file resnet18_quant.py, function quantization, this line of code is for resnet18 on imagenet:
`input = torch.randn([batch_size, 3, 224, 224])`
It should be changed to your own model input shape (it seems that [1, 3, 640, 640]).

closed 2023-03-02 by niuxjxlnx

IkrameBeggar, 2023-09-21:
> Hi!iI have some questions to ask you.This error occurs when i run this commond"python resnet18_quant.py --quant_model float" to quantize yolov5 model(use the code of "ultralytics"): [image] Have you [...]

Hello can you please share with me the files in my email address: ikrame.bgr@gmail.com