Xilinx/Vitis-AI #1079: Parameter tensor is not quantized for additional parameter

j0r0123, 2022-11-01:
I want to implement my PyTorch model, but the quantizer is not able to find and quantize all the parameters. Here is my setting:
- Docker Image Version: 2.5.0.1260 (CPU)
- Vitis AI Git Hash: 502703c
- Pytorch Workflow: conda activate vitis-ai-pytorch 

My model requires an extra parameter that is used as multiplicator and looks like this (reduced example, for the sake of simplicity): 
```
class myModel(nn.Module):
    def __init__(self):
        super(myModel, self).__init__()
        self.myParameter = nn.Parameter(torch.ones([1, 3, 360, 640]))

    def forward(self, x: torch.Tensor):
        return x*self.myParameter
```

My Quantization Script looks like this:
```
dtype = torch.float32
device = torch.device('cpu')
model = myModel().to(dtype=dtype, device=device)
model.load_state_dict(torch.load("weights.pt"))

rand_in = torch.randn([batchsize, 3, 360, 640])
input = ... some code to load data ...

quantizer = torch_quantizer(quant_mode, model, rand_in, output_dir=quant_model)
quantized_model = quantizer.quant_model

quantized_model.eval()
quantized_model.to(device)
output = quantized_model(input)

# export config
if quant_mode == 'calib':
    quantizer.export_quant_config()
if quant_mode == 'test':
    quantizer.export_xmodel(deploy_check=False, output_dir=quant_model)
```
And the error is the following:
```
[VAIQ_WARN]: CUDA is not available, change device to CPU

[VAIQ_NOTE]: Quant config file is empty, use default quant configuration

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cpu'.

[VAIQ_NOTE]: =>Parsing myModel...

[VAIQ_NOTE]: Start to trace model...

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 3/3 [00:00<00:00, 1853.15it/s, OpInfo: name = return_0, type = Return]

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(build/quant_model/myModel.py)

[VAIQ_NOTE]: =>Get module with quantization.

[VAIQ_NOTE]: =>Exporting quant config.(build/quant_model/quant_info.json)

[VAIQ_WARN]: Parameter tensor is not quantized: myModel::myParameter

[VAIQ_WARN]: Quantization is not performed completely, check if model inference function is called!!!
```

For the sake of completeness, the generated Python model and the quantization information look like this:
```
class myModel(torch.nn.Module):
    def __init__(self):
        super(myModel, self).__init__()
        self.module_0 = py_nndct.nn.Input() #myModel::input_0
        self.module_1 = py_nndct.nn.Module('elemwise_mul') #myModel::myModel/8
        self.myParameter = torch.nn.parameter.Parameter(torch.Tensor(1, 360, 640, 3))

    def forward(self, *args):
        output_module_0 = self.module_0(input=args[0])
        output_module_0 = self.module_1(input=output_module_0, other=self.myParameter)
        return output_module_0
```
```
{
  "param":
  {
    "myModel::myParameter":[8,null]
  },
  "output":
  {
    "myModel::input_0":[8,4],
    "myModel::myModel/8":[8,4]
  },
  "input":
  {

  },
  "fast_finetuned":false,
  "bias_corrected":true
}
```

Note that all remaining parameter and the outputs of the model were successfully quantized. 

I had a look at the code and the problem seems to be that during the evaluation step the quantizer only checks the input tensors of a module against the input list. Thereby it misses, that the input could also be an additional parameter inside the module. 

Is there a fix or a workaround for this problem?

hoseong-kim, 2022-11-10:
I have a same issue. Any solution?

niuxjxlnx, 2022-11-14:
@j0r0123 ，

In Vitis-AI 2.5, it does not support independent parameter quantization. This is expected and depends on our IP design, which only has optimization on parameter in conv/linear/norm.

Nabih-Saleh, 2023-09-05:
same problem!  anybody got an idea?