Xilinx/Vitis-AI #1080: Algorithm of convolution

WhiteCri, 2022-11-02:
Hello, 

I would like to look around performance and algorithm of convolution layer computation.
Can you provide us manual or resource for stuyding?

Thanks,

wangxd-xlnx, 2022-11-02:
Hi @WhiteCri 

You could refer to this blog, it's concise and clear.
https://machinelearningmastery.com/convolutional-layers-for-deep-learning-neural-networks/

WhiteCri, 2022-11-05:
Thank you for kind introduction, but I know how to calculate convolution layer. I think my question was not clear.

I want to learn how vitis-ai integrate data-reuse, pipeline, loop tiling or etc into convolution layer computation. Can you provide some concrete explanations?

wangxd-xlnx, 2022-11-07:
Hi @WhiteCri 

I still don't quite understand what you're asking. Do you want to know how it is implemented on the dpu core?

WhiteCri, 2022-11-07:
Oh yeah that's correct! Can you provide some introductions?

qianglin-xlnx, 2023-01-17:
Hi @WhiteCri 
Sorry. Currently. Vitis-ai compiler is not open source. So we can't provide more information about it.

closed 2023-01-17 by qianglin-xlnx