Xilinx/Vitis-AI #1086: no mc_code after converting pytorch model to xmodel

eerobert, 2022-11-14:
Already submitted in Xilinx forum, wish to have more comments from here

https://support.xilinx.com/s/question/0D54U00005ranI3SAI/vart-result-mismatch?language=en_US

qianglin-xlnx, 2022-11-14:
Hi @eerobert 
1 Which Vitis AI version do you use? Can you run a basic model, such as resnet50_tf, successfully on your kv260 board? 
2 When you say "converting pytorch model to xmodel" , do you mean quantizing the model or compiling the model?
If possile, can you share the quantized model and the compiled model. Thank you.

eerobert, 2022-11-15:
VAI2.5

Yes, I can run resnet and segmentation successfully with models from model-zoo and recompiled to B3136

I can quantized the model and the evaluation (value from loss function) from pytorch looks good.
But after compilation to B3136 and run from loadapp kv260-smartcam, the result mismatch.

eerobert, 2022-11-15:
[deepWBNet_int.zip](https://github.com/Xilinx/Vitis-AI/files/10009740/deepWBNet_int.zip)
[deepWBNet.zip](https://github.com/Xilinx/Vitis-AI/files/10009741/deepWBNet.zip)

deepWBNet_int is the quantized model
deepWBNet is the compiled quantized model (B3136)

qianglin-xlnx, 2022-11-16:
Hi @eerobert 
Thank you for sharing the model. 
1 For the "no mc_code" message, it's a warning message. Please ignore it.
2 For the error you encounter, you can switch to the root, then run the xdputil again.
   `sudo -s`

eerobert, 2022-11-17:
thx for your comments and things seem to get a bit progress

now i am facing another error, complaining no such file

//
FileNotFoundError: [Errno 2] No such file or directory: './deepWBNet/input_aquant.bin'
fillin deepWBNet__input_0_fix
//

it seems like there are such a file if it is tensorflow flow for debugging.. (according to UG)
however, i can't find such a file if i run my model from pytorch

and just want to recap, my greater goal is to deploy my model on kv260. not only solving these glitches during debugging

thanks

eerobert, 2022-11-17:
thx for your comment and seems like i have a bit progress.

however, when i run the xdputil to debug the model running on kv260, i now have the following error.

//
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1117 00:57:24.248212  6074 dpu_kernel.cpp:178] child_name subgraph_deepWBNet__deepWBNet_UpBlock_decoder_up1__input_41 has no mc_code
F1117 00:57:24.301489  6074 device_memory_edge.cpp:82] Check failed: fromdata != ((void *) -1) (0xffffffffffffffff vs. 0xffffffffffffffff) fd = 8 base=0x361f0000
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:  6074 Aborted                 (core dumped) /usr/bin/python3 -m xdputil $*
//

still i cannot generate the comparison between VART tools result with kv260 dpu results...
and my greater goal is to have the model running on kv260... seems still very far away from it. 

i also attach the quantizer.export_xmodel(deploy_check=True) result here for your reference

[deploy_check_data_int.zip](https://github.com/Xilinx/Vitis-AI/files/10026817/deploy_check_data_int.zip)

eerobert, 2022-11-22:
@qianglin-xlnx 

any update on this case?

eerobert, 2022-11-22:
Hi @qianglin-xlnx 

What command u actually run for xdputil on the model?
I was following the UG1414 - DPU Debug with VART to crosscheck the DPU result with VART result

I already attached my DPU result in previous post. 
However, when I try to xdputil run my model to do the crosscheck, it crashed.

qianglin-xlnx, 2022-11-22:
`xilinx-kv260-starterkit-20221:~$ env XLNX_ENABLE_DUMP=1 XLNX_ENABLE_DEBUG_MODE=1  xdputil run deepWBNet2.xmodel deepWBNet2.xmodel -i 1
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1121 22:14:46.620813 212029 dpu_kernel.cpp:178] child_name subgraph_deepWBNet__deepWBNet_UpBlock_decoder_up1__input_41 has no mc_code
fillin deepWBNet__input_0_fix
F1121 22:14:46.659126 212029 device_memory_edge.cpp:79] Check failed: fd > 0 (-1 vs. 0)
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20: 212029 Aborted                 /usr/bin/python3 -m xdputil $*
xilinx-kv260-starterkit-20221:~$ sudo -s
Password:
xilinx-kv260-starterkit-20221:/home/petalinux# env XLNX_ENABLE_DUMP=1 XLNX_ENABLE_DEBUG_MODE=1  xdputil run deepWBNet2.xmodel deepWBNet2.xmodel -i 1
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1121 22:15:15.760076 212207 dpu_kernel.cpp:178] child_name subgraph_deepWBNet__deepWBNet_UpBlock_decoder_up1__input_41 has no mc_code
fillin deepWBNet__input_0_fix
I1121 22:15:15.799966 212207 dpu_runner_base_imp.cpp:331] dump   to dump/subgraph_deepWBNet__deepWBNet_DoubleConvBlock_encoder_inc__Sequential_double_conv__Conv2d_0__input_3/input/0.deepWBNet__input_0_fix_upload_0.bin device_core_id 0 reg_id 2 base 1062666240 tensor_offset 400 offset 1062666640 tensor_size 49152 dump_ok success
I1121 22:15:15.805872 212207 dpu_runner_base_imp.cpp:331] dump   to dump/subgraph_deepWBNet__deepWBNet_DoubleConvBlock_encoder_inc__Sequential_double_conv__Conv2d_0__input_3/output/0.deepWBNet__deepWBNet_DoubleConvBlock_encoder_inc__Sequential_double_conv__ReLU_1__input_5_fix.bin device_core_id 0 reg_id 1 base 1149239296 tensor_offset 98304 offset 1149337600 tensor_size 393216 dump_ok success
I1121 22:15:15.811185 212207 dpu_runner_base_imp.cpp:331] dump   to dump/subgraph_deepWBNet__deepWBNet_DoubleConvBlock_encoder_inc__Sequential_double_conv__Conv2d_2__input_7/input/0.deepWBNet__deepWBNet_DoubleConvBlock_encoder_inc__Sequential_double_conv__ReLU_1__input_5_fix.bin device_core_id 0 reg_id 1 base 1149239296 tensor_offset 98304 offset 1149337600 tensor_size 393216 dump_ok success
I1121 22:15:15.817052 212207 dpu_runner_base_imp.cpp:331] dump   to dump/subgraph_deepWBNet__deepWBNet_DoubleConvBlock_encoder_inc__Sequential_double_conv__Conv2d_2__input_7/output/0.deepWBNet__deepWBNet_DoubleConvBlock_encoder_inc__Sequential_double_conv__ReLU_3__2714_fix.bin device_core_id 0 reg_id 1 base 1149239296 tensor_offset 491520 offset 1149730816 tensor_size 393216 dump_ok success
`

eerobert, 2022-11-22:
it looks like you are using oob startkit sd image. am i correct?

eerobert, 2022-11-22:
hi

i can execute //sudo xdputil run deepWBNet.xmodel ./golden_path/deeWBNet__input_0_fix.bin //

but cannot run any of the following

//sudo env XLNX_GOLDEN_DIR=./path/ xdputil run deepWBNet.xmodel ./golden_path/deeWBNet__input_0_fix.bin //
//sudo env XLNX_ENABLE_DUMP=1 xdputil run deepWBNet.xmodel  ./golden_path/deepWBNet__input_0_fix.bin //

qianglin-xlnx, 2022-11-22:
Hi @eerobert 
I'm sorry, I still can't reproduce the issue you encounter. So which kv260 image do you use?

[image]

[image]

eerobert, 2022-11-23:
https://ubuntu.com/download/amd-xilinx

Ubuntu Desktop 22.04 LTS

And what is your image?

qianglin-xlnx, 2022-11-23:
I used the petalinux system image not ubuntu Desktop 22.04 LTS.

eerobert, 2022-11-23:
will you look into the problem using ubuntu image?

qianglin-xlnx, 2022-11-23:
Hi @eerobert 
I run your compiled model which is b3136 on kv260.
I also recompiled your xmodel with b4096 and run it on zcu102. Both are ok.
Since FAE gets the same issue feedback from Avnet and trying to test the ubuntu image, I will not going to try the ubuntu image. Thank you.

closed 2022-11-24 by eerobert