Xilinx/Vitis-AI #1087: The type of attr 'end' should be <class 'int'> instead of <class 'nndct_shared.nndct_graph.base_tensor.Tensor'>

aris-t, 2022-11-14:
**Issue:**
Hi Folks, 
I am trying to quantize a U net neural network and running into the above error. The quantization script is based on the standard PyTorch quantization and the U net based on an online example. 

**Starting Sources:**
https://amaarora.github.io/2020/09/13/unet.html
https://github.com/Xilinx/Vitis-AI-Tutorials/blob/1.4/Design_Tutorials/09-mnist_pyt/README.md

**Full Trackback**
```
GO ENCODER
GO DECODER
PyTorch version :  1.10.1
3.7.12 | packaged by conda-forge | (default, Oct 26 2021, 06:08:53) 
[GCC 9.4.0]
 Command line options:
--build_dir    :  ./build
--quant_mode   :  calib
--batchsize    :  100
You have 2 CUDA devices available
 Device 0 :  Quadro RTX 6000
 Device 1 :  Quadro RTX 6000
Selecting device 0..
Loading: ./build/float_model/f_model.pth
Putting to: cuda:0

[VAIQ_NOTE]: Quant config file is empty, use default quant configuration

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing UNet...

[VAIQ_NOTE]: Start to trace model...

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops... 
Traceback (most recent call last):
  File "quantize.py", line 130, in <module>
    run_main()
  File "quantize.py", line 123, in run_main
    quantize(args.build_dir,args.quant_mode,args.batchsize)
  File "quantize.py", line 79, in quantize
    quantizer = torch_quantizer(quant_mode, model, (rand_in), output_dir=quant_model)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/apis.py", line 98, in __init__
    quant_config_file = quant_config_file)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/base.py", line 151, in __init__
    device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/utils.py", line 193, in prepare_quantizable_module
    graph = parse_module(module, input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/utils.py", line 83, in parse_module
    module, input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/parse/parser.py", line 74, in __call__
    nndct_graph = self._convert_graph(raw_graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/parse/parser.py", line 121, in _convert_graph
    nndct_node = self._convert_node(raw_node)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/parse/parser.py", line 211, in _convert_node
    nndct_node.op = self._create_op(raw_node.kind, nndct_node, node_input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/parse/parser.py", line 244, in _create_op
    op = getattr(op_creator, op_type)(*node_input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/parse/op_dispatcher.py", line 624, in strided_slice
    op.set_config("end", end)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_operator.py", line 450, in set_config
    setattr(self, config_name, value)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/parse/torch_op_def.py", line 1388, in end
    self.set_attr(self.AttrName.END, end)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_operator.py", line 399, in set_attr
    self._attrs[attr_name].value = attr_value
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_operator.py", line 165, in value
    f"The type of attr '{self._name.value}' should be {self._type} instead of {type(value[0])}"
TypeError: The type of attr 'end' should be <class 'int'> instead of <class 'nndct_shared.nndct_graph.base_tensor.Tensor'>
```

**Requests:**
I am not really sure how to go about fixing this issue. The few searches I have done have yielded minimal results. If needed I can post code and other information. Just let me know what is worth posting.

carrotandjzy, 2022-11-17:
So far, we just support slice operation with constant index. We'll fix this issue in vai 3.0.

aris-t, 2022-11-18:
Thank you, 
Would you mind expanding on how you went about isolating the issue? That is one of the main things I am trying to learn about the vitis tools, how to debug and understand what is failing so that I may fix it on my own in the future.

carrotandjzy, 2023-02-28:
Hi @aris-t 
Compared with 2.5,  attribute "end" of slice support both int and tensor types in vai3.0 
https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_quantizer/vai_q_pytorch/nndct_shared/nndct_graph/operator_definition.py#L1150

closed 2023-06-26 by niuxjxlnx