Xilinx/Vitis-AI #1088: Quantization error for vai_q_tensorflow2 : ValueError: Cannot assign value to variable: Shape mismatch.

alpereenkaratas, 2022-11-16:
I had previously trained a retinanet model locally. I am trying to quantize this model using vai_q_tensorflow2 with Vitis AI 2.5. A code block for quantization is as follows:

`
float_model = tf.keras.models.load_model('snapshots/inference.h5', compile=False, custom_objects=custom_objects)
quantizer = vitis_quantize.VitisQuantizer(float_model)
`
`
quantized_model = quantizer.quantize_model(calib_dataset=calib_dataset, add_shape_info=True) # gives error here
`

The error I get is shown below.

`Traceback (most recent call last):
  File "quantize.py", line 177, in <module>
    main()
  File "quantize.py", line 173, in main
    quant_model(args.float_model, args.quant_model, args.batchsize, args.predict, args.pred_dir)
  File "quantize.py", line 116, in quant_model
    quantized_model = quantizer.quantize_model(calib_dataset=calib_dataset, add_shape_info=True)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 841, in quantize_model
    self.optimize_model()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 648, in optimize_model
    self._create_optimized_model()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 362, in _create_optimized_model
    quantize_strategy=self._quantize_strategy)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 1004, in create_optimize_model
    model, candidate_layers, layer_metadata)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 82, in apply
    model, configs, available_transforms, candidate_layers, layer_metadata)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 54, in _apply_availables
    layer_metadata).recursive_transform()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 740, in recursive_transform
    self.layer_metadata).transform()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 706, in transform
    self._set_layer_weights(layer, weights_map)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 595, in _set_layer_weights
    K.batch_set_value(weight_value_tuples)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow/python/util/traceback_utils.py", line 153, in error_handler
    raise e.with_traceback(filtered_tb) from None
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/backend.py", line 4019, in batch_set_value
    x.assign(np.asarray(value, dtype=dtype_numpy(x)))
ValueError: Cannot assign value to variable ' pyramid_regression_0/kernel:0': Shape mismatch.The variable shape (3, 3, 256, 256), and the assigned value shape (3, 3, 256, 36) are incompatible.
`

I don't know what to do about solving this error. I am sharing my model with the name 'inference.h5' in the link:
https://drive.google.com/file/d/1q_5uZUeR_7J8K6UVnK-dV68qG0fBxV0J/view?usp=share_link

amirversal, 2022-11-25:
Hi, could you solve the above issue?

alpereenkaratas, 2022-12-04:
Hi @amirversal. Unfortunately i couldn't solve it. For the details, please check #951.

gengxinwu, 2023-02-15:
Hi @alpereenkaratas. The retinanet model contains two reused sub-models that vai_q_tensorflow2 does not support currently. For a workaround, we can flatten the sub-models and make several copies depending on reused times.

I will explain the modifications in detail referring to [this repo](https://github.com/fizyr/keras-retinanet) ：

1. Modify the [__build_model_pyramid](https://github.com/fizyr/keras-retinanet/blob/main/keras_retinanet/models/retinanet.py#L202) function as below
```
def __build_model_pyramid(name, model, features):
    """ Applies a single submodel to each FPN level.

    Args
        name     : Name of the submodel.
        model    : The submodel to evaluate.
        features : The FPN features.

    Returns
        A tensor containing the response from the submodel on the FPN features.
    """
    def _clone_layer_fn():
        def __clone_fn(layer):
            cloned_layer = layer.__class__.from_config(layer.get_config())
            cloned_layer._name += "_"  # this avoids layer name duplication
            return cloned_layer
        return __clone_fn

    model0 = model
    model1 = keras.models.clone_model(model0, clone_function=_clone_layer_fn())
    model2 = keras.models.clone_model(model1, clone_function=_clone_layer_fn())
    model3 = keras.models.clone_model(model2, clone_function=_clone_layer_fn())
    model4 = keras.models.clone_model(model3, clone_function=_clone_layer_fn())
    P3 = model0.call(features[0])
    P4 = model1.call(features[1])
    P5 = model2.call(features[2])
    P6 = model3.call(features[3])
    P7 = model4.call(features[4])
    return keras.layers.Concatenate(axis=1, name=name)([P3, P4, P5, P6, P7])
```
2. Create a new model with original weights before quantization
```
with custom_object_scope(custom_objects):
    float_model = tf.keras.models.load_model('snapshots/inference.h5', compile=False, custom_objects=custom_objects) 

    from keras_retinanet.models.resnet import resnet50_retinanet
    resnet_retinanet = resnet50_retinanet(num_classes=4)
    from keras_retinanet.models.retinanet import retinanet_bbox
    model = retinanet_bbox(resnet_retinanet)

    for layer in float_model.layers:
        if isinstance(layer, keras.Model):
            for sublayer in layer.layers:
                if sublayer.__class__.__name__ == "InputLayer":
                    continue  # skip submodel's input

                extracted_weights = sublayer.get_weights()
                layer_basename = sublayer.name
                for i in range(5):
                    model.get_layer(layer_basename).set_weights(extracted_weights)
                    layer_basename += '_'  # this discriminates the 5 copies of submodels
        else:
            model.get_layer(layer.name).set_weights(layer.get_weights())

    model.save('./snapshots/inference_flattened.h5')
```

Now the new saved inference_flattened.h5 can be loaded again and fed into the quantizer.

**Notes**
When quantizing the new inference_flattened.h5, you may encounter a new error related to _BatchNormalization_. This is because the _BatchNormalization_ layer in the model comes from customized keras_resnet.layers._batch_normalization.BatchNormalization instead of standard tf.keras.layers. BatchNormalization.
So we can switch them back:
```
float_model = tf.keras.models.load_model('./snapshots/inference_flattened.h5', compile=False, custom_objects=custom_objects)
clone_fn = lambda layer : keras.layers.BatchNormalization(name = layer.name, weights = layer.get_weights()) \
                              if layer.__class__.__name__ == "BatchNormalization" else layer
cloned_model = keras.models.clone_model(float_model, clone_function=clone_fn)
cloned_model.save('./snapshots/inference_flattened_replaced.h5')
```

renjiangyong, 2023-06-20:
This issue has been inactive for 4 months, so we will close it for now. If you have any further concerns or need assistance, please feel free to reopen it at any time.

closed 2023-06-20 by renjiangyong