Xilinx/Vitis-AI #1091: xmodel file of YOLOX

aiqiangbrave, 2022-11-19:
**### _Could you help me analyze the size of the model? I remember that the xmodel model after yolov4 compression is about 7M, but the model after yolox compression is only twice shrunk (about 90M), is this normal? Thank you so much_**

aiqiangbrave, 2022-11-21:
> Could you help me analyze the size of the model? I remember that the xmodel model after yolov4 QAT is about 7M, but the model after yolox QAT is only twice shrunk (about 100M), is this normal? Why is the quantified model still so large？ [image] [image]
> 
> Thank you so much~~~

I used the run_train.sh trained float model to generate the xmodel file, and although it was successful, I compiled it with the following error:
**Instruction format:**
vai_c_xir -x /PATH/TO/quantized.xmodel -a /PATH/TO/arch.json -o /OUTPUTPATH 
-n netname
**error:**
[UNILOG][FATAL][XCOM_PM_FAIL][The compiler occurs an error when generating instructions, please contact us.]

qianglin-xlnx, 2023-02-20:
Closing since no activity for more than 1 month, please reopen if you still have any questions, thanks

closed 2023-02-20 by qianglin-xlnx