Xilinx/Vitis-AI #1099: Unable to quantize model in tensorflow2 environment

davendramaharaj1, 2022-11-30:
I lauched the docker image xilinx/vitis-ai-gpu:2.5 with the following command:

`docker run -it --gpus all xilinx/vitis-ai-gpu:2.5 nvidia-smi
`
I created a yolov4-tf2 environment that uses python 3.9. From there, I installed all the requirements in tf_yolo4_coco_416_416_60.3G_2.5/code/requirements but changed to tensorflow==2.8 since I have an RTX3000 series card.

I then ran conda activate tensorflow2. I downloaded the data set, converted the model to tensorflow (.pb) format using python tools/model_converter/convert.py and python tools/model_converter/keras_to_tensorflow.py

From here, I entered tf_yolo4_coco_416_416_60.3G_2.5/code/quantized. I changed bash quantized_416 to run vai_q_tensorflow2 since I am using tensorflow2 environment. I get the following:

[image]

I am not sure why vai_q_tensorflow2 doesn't exist but I tried building it. First, I tried the "nstall from Source Code with the Wheel Package" method and ended up with the following:

[image]

I am very confused on what's going on...please help

AbdoSaeed, 2022-12-03:
Hi @davendramaharaj1 
I have been in the same situation for a week. I have not fully resolved my issue yet. but here are some remarks that may help

1. It looks like there is no bash command vai_q_tensorflow2. Instead, there is a python package that provides the functions to do quantization and inspection. 
Check the examples from the vitis-ai user guide https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Quantizing-Using-the-vai_q_tensorflow2-API
2. I have managed to install vai_q_tensorflow2.x with the following steps
`cd src/Vitis-AI-Quantizer/vai_q_tensorflow2.x
export SRC_DIR=$PWD; ## or setenv SRC_DIR=$PWD in csh
sh vai_q_tensorflow2_cpu_feedstock/build.sh`
3. After step 2, python package tensorflow_model_optimization got recognized but I got another error that tensorflow is not defined. I am still not sure how to proceed after that. Please let me know if was able to solve this or if you find any useful info

AbdoSaeed, 2022-12-04:
I managed to make it work with
`pip install dm-tree~=0.1.1`
`pip install tensorflow==2.8.0`
Not sure if other tensorflow versions can work correctly or not. But this what I tried and it works.

AbdoSaeed, 2022-12-04:
I managed to make it work after installing the following
`pip install dm-tree~=0.1.1`
`pip install tensorflow==2.8.0`
Not sure if other tensorflow versions can work or not.

zhenzhen-AMD, 2023-06-14:
Hi @davendramaharaj1  and @AbdoSaeed ,
1. The "vai_q_tensorflow2" instruction does not exist.
2. For the "tf1.pb" model, please use the docker environment: conda activate vitis-ai-tensorflow.
3. For the "tf2.h5" model, please use the docker environment: conda activate vitis-ai-tensorflow2.
4. To install "vai_q_tensorflow2" from source in Docker 2.5, you will need Python 3.7. Follow these instructions:
    - pip install -r requirements.txt
    - bash ./pip_pkg.sh ./pkgs/ --release
    - pip install pkgs/*.whl
5. To install "vai_q_tensorflow2" from source in a local environment, you will need Python 3.7. Follow these instructions:
    - pip install -r requirements.txt
    - sh build.sh
    - pip install pkgs/*.whl
6. In vitis-ai-2.5, the highest supported version of TensorFlow is 2.8. In the upcoming vitis-ai-3.5 release, the highest supported version of TensorFlow will be 2.12.

I hope this helps! Let me know if you have any further questions.

zhenzhen-AMD, 2023-06-26:
Hi @davendramaharaj1 and @AbdoSaeed ,
This issue has been open for a while, and we will now close it. If you have any future questions, feel free to open a new issue. Thank you for your understanding.

closed 2023-06-26 by qianglin-xlnx