Xilinx/Vitis-AI #1101: No dpu_trd BSP or DPU kernel module

Edocit, 2022-12-02:
Hi I was using Vitis AI 2.0 on ZCU 102 and ZCU 104 but in this version there is no reference zcu102trd.bsp or the DPU Kernel module. How to integrate for a Petalinux build for the Kria KR260. I'm using Petalinux 2022.2, Vivado 2022.2, and Vitis 2022.2. Thank you

yangye8, 2023-01-30:
Please refer to this page https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/1641152513/Kria+K26+SOM  to get building instructions for KR260.  SOM has a separate release page https://github.com/Xilinx/kria-vitis-platforms which is not included in Vitis AI GitHub

qianglin-xlnx, 2023-03-27:
Closing since no activity for more than 1 month, please reopen if you still have any questions, thanks

closed 2023-03-27 by qianglin-xlnx