Xilinx/Vitis-AI #1105: tensorflow VisionTransformer compile error

wang7393, 2022-12-06:
I use this model in Xilinx model zoo do quantize and compile for zcu102:
[image]

the freeze and quantize model is as followings:
link：https://pan.baidu.com/s/1H0MShq2cQDmiYUNtoYI-hg 
code：x2k7

The compile command is as follows:
`vai_c_tensorflow -f quantize_results/quantize_eval_model_1000.pb    -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json     -o quantize_results/ -n vit  --options '{"input_shape": "1,352,3521,3"}'`

and get the following error:
[image]

print the self.tmp_params:
`{'Tmultiples': 3, 'data_type': 'float32', 'name': 'Tile', 'kind': 'Tile'}`

hocken-li, 2023-05-26:
@wang7393 Hello!
I am also implementing VIT recently from model zoo. I get some problem, could you help me?
There is no ckpt.meta in the folder, and I can't freeze the model. How do you make it？
Or would you like to share the freeze and quantize model to me? The link is no longer valid.
It will be very happy if I can add your wechat to learn more! 
Thanks!

wangxd-xlnx, 2023-06-14:
Hi @wang7393 

Sorry for the late reply. This ViT quantized model need to match customized DPU and xclbin. You could refer to this demo.
https://github.com/Xilinx/Vitis-AI/tree/2.5/examples/Transformer