Xilinx/Vitis-AI #1106: TensorFlow2 - QAT - yolov4-tiny - high loss

hito0512, 2022-12-06:
using tensorflow2 QAT function to quant the yolov4-tiny model, but the finetune loss is too high , the normal loss(training float model) like follow pic
[image]
,but the qat loss is as follow 
[image]
the reason is the location_loss  become too high。 loss= location_loss + class_loss + confidence_loss 。
Is there anything else I need to set up?

jimheaton, 2022-12-06:
What learning rate are you using? The learning rate  typically this needs to be set much smaller for QAT than regular training.

hito0512, 2022-12-09:
@jimheaton 
the learning rate  is 1e-4.This has nothing to do with the learning rate. 
I printed the loss of two outputs，and the result is as follow

`15/360 [>.............................] - ETA: 1:34 - loss: 28.3271 - quant_conv2d_17_loss: 0.6805 - quant_conv2d_20_loss: 6.8247 16/360 [>.............................] - ETA: 1:34 - loss: 28.3311 - quant_conv2d_17_loss: 0.6846 - quant_conv2d_20_loss: 6.8502 17/360 [>.............................] - ETA: 1:34 - loss: 28.3269 - quant_conv2d_17_loss: 0.6806 - quant_conv2d_20_loss: 6.8882 18/360 [>.............................] - ETA: 1:34 - loss: 28.3303 - quant_conv2d_17_loss: 0.6841 - quant_conv2d_20_loss: 6.9084 19/360 [>.............................] - ETA: 1:35 - loss: 28.3309 - quant_conv2d_17_loss: 0.6849 - quant_conv2d_20_loss: 6.9302 20/360 [>.............................] - ETA: 1:35 - loss: 28.3278 - quant_conv2d_17_loss: 0.6819 - quant_conv2d_20_loss: 6.9384 21/360 [>.............................] - ETA: 1:35 - loss: 28.3287 - quant_conv2d_17_loss: 0.6830 - quant_conv2d_20_loss: 6.9408`

the total loss is not equal quant_conv2d_20_loss + quant_conv2d_17_loss, the result is similar with the first picture. i found a  [Similarity problem](https://stackoverflow.com/questions/55969404/summing-multiple-loss-in-keras#comment98594668_55971032). it seems regularizer loss causing this .
Although the loss was great, I tested the QAT model，the result was good。
Do I need to pay attention to the loss of regularizer?

Thank you for your reply

renjiangyong, 2023-09-14:
Closing since no activity for more than 2 weeks, please reopen if you still have question, thanks

closed 2023-09-14 by renjiangyong