Xilinx/Vitis-AI #1108: vai_c_xir PyTorch compilation fails with XCOM_ITEM_UNDEFINED

afzalxo, 2022-12-08:
I am using U50 with Vitis AI 1.4.1.978 and `DPUCAHX8L_ISA0` to run and evaluate thousands of custom convolutional models to construct a large dataset of convnet architectures along with their performance on board. 

For some models, when I try to compile the quantized model using vai_c_xir, the compilation fails with 
```
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][XCOM_ITEM_UNDEFINED][The requested item is not finded or defined!]
*** Check failure stack trace: ***
Aborted (core dumped)
```
A sample quantized xmodel file is here: [custom_convnet.xmodel.zip](https://github.com/Xilinx/Vitis-AI/files/10181697/custom_convnet.xmodel.zip).

Help would be appreciated. Thanks

afzalxo, 2022-12-08:
I have identified many different models that are valid, can be trained and quantized, but do not compile using `vai_c_xir` due to `XCOM_ITEM_UNDEFINED` error. I cannot however make out a pattern; different expansion factors in MobileNet-like models, and different number of layers per block also generate this result. For some expansion factors/number of layers, things work fine, but fail for others without a proper justification of failures. The compiler should at least throw a decipherable error message, which is not the case here.

afzalxo, 2022-12-08:
A few updates:
I compiled the same quantized model using ZCU102 as the target for `DPUCZDX8G` and Vitis AI 2.5 and the compilation goes smoothly. The error only appears when compiling for DPU `DPUCAHX8L` utilized by U50 (Can't use `DPUCAHX8H` since I'm using depthwise conv). The operations contained in the model are all supported by both DPUs. Hence the problem is likely on the compilers end `vai_c_xir`.

SuZhan0322, 2023-06-13:
Hi @afzalxo , Is your model a pytorch model? And the vai_c_xir only supports the compilation of pytorch models. Thanks!