Xilinx/Vitis-AI #1110: Model Zoo, Multi-threaded Performance

afzalxo, 2022-12-11:
The model zoo lists the performance of different models in both single threaded and multi-threaded execution. What is the thread count utilized to obtain the results from multi-threaded configuration? The [README.md](https://github.com/Xilinx/Vitis-AI/tree/master/model_zoo) does not list the thread count utilized for evaluating the FPS numbers for any of the FPGAs.

Thanks

afzalxo, 2022-12-11:
Could you kindly also specify the number of DPU cores in the DPU IP `DPUCZDX8G_ISA1_B4096` by default? It should be 2 cores if I'm not wrong.

afzalxo, 2022-12-12:
> Could you kindly also specify the number of DPU cores in the DPU IP `DPUCZDX8G_ISA1_B4096` by default? It should be 2 cores if I'm not wrong.

Its 3 cores.

xihuazhanghui, 2023-01-16:
If you use the official released images, you can use tool ` show_dpu ` to get the board dpu info.

afzalxo, 2023-01-16:
This works, thanks a lot. Also, is there any way to check the power consumption of the chip? Save for attaching an external power meter.

xihuazhanghui, 2023-01-17:
This feature did not release yet.

afzalxo, 2023-01-31:
Thanks. Closing the thread for now.

closed 2023-01-31 by afzalxo