Xilinx/Vitis-AI #1111: DPU Peak Power Consumption ZCU102

afzalxo, 2022-12-11:
I am using `DPUCZDX8G` with ZCU102 and Vitis AI 2.5 to implement a custom convolutional model. The model works fine on the board and I am able to execute and obtain metrics such as throughput and latency. The board has petalinux installed but the `xbutil` utility does not show the power consumption of the board. I also tried `vaitrace_py` by adding `power_profile=true` to the `xrt.ini` file but the power profile csv thats supposed to be generated isn't generated by the `vaitrace_py` utility, which makes me believe that power profile feature is not supported by this FPGA board.

Is there any other way of measuring the power consumption of my design, save for connecting a power meter to the board?
Thanks

jileilxinlix, 2023-01-18:
Hi @afzalxo , 
Maybe you could run power_mon.py to get some power consumption info, which file in /usr/bin/xlnx/vaitrace/vaitraceTools.

afzalxo, 2023-01-18:
Hi @jileilxinlix, File `power_mon.py` does not exist in `/usr/bin/xlnx/vaitrace/vaitraceTools`. I am using the official board image v2.5.0 from [here](/usr/bin/xlnx/vaitrace/vaitraceTools) for ZCU102. The link to download the board image was taken from [here](https://github.com/Xilinx/Vitis-AI/tree/2.5/setup/mpsoc).

The files in the dir are as follows:
[image]

jileilxinlix, 2023-01-18:
Hi @afzalxo ,
Maybe  you could try to install  all *.rpm in the board manual, and the rpm package link: https://github.com/Xilinx/Vitis-AI/tree/2.5/setup/mpsoc/2022.1

jileilxinlix, 2023-01-30:
@afzalxo ,I am sorry give you a mistake answer.
After double check your question , I'am sorry to tell you that ZCU102 has no this function.
You could check the code which in :https://github.com/Xilinx/XRT/blob/991d7942717232de3723ddde71e211653b5852d4/src/runtime_src/core/edge/user/plugin/xdp/plugin_loader.cpp#L63

afzalxo, 2023-01-31:
No worries. Thanks for the help.

closed 2023-01-31 by afzalxo

AmirMahmood, 2023-05-10:
Same problem here. did you find any solution?
@afzalxo

afzalxo, 2023-05-11:
Hi, the ZCU102 does not have the system controller that allows measuring power. So to my knowledge, the only solution is to use an external power meter.