Xilinx/Vitis-AI #1112: Throughput decreases when # threads is a multiple of 3

afzalxo, 2022-12-12:
I am evaluating DPUCZDX8G_ISA1_B4096 on ZCU102 Vitis AI 2.5. When I use # threads which is a multiple of 3, the performance decreases. The issue can be replicated using `resnet_mt_py` example in VART offered [here](https://github.com/Xilinx/Vitis-AI/tree/master/examples/VART/resnet50_mt_py). I also evaluated multiple custom models using different # threads in range [1,9] and obtained the same result. Whenever I use # threads = 3, 6, or 9, the performance sees a drop. I'm assuming it was to do with the fact that this DPU IP uses 3 DPU cores. I noticed on VAI profiler that DPU 3 always performs the worst and DPU 1 always performs the best in terms of throughput.

The following plot shows the performance of several custom models (on primary vertical axis), and `resnet50_mt_py` in red color (on secondary vertical axis), when evaluated using different number of threads.

[image]

yangyecs, 2023-06-26:
Hi afzalxo
Assuming you are using the ZCU102 board with Xilinx provided 3 dpu cores image.  In your statement, you mentioned  "DPU 3 always performs the worst and DPU 1 always performs the best in terms of throughput", which aligns with expected result.  For more details, please refer to this [hpc](https://github.com/Xilinx/Vitis-AI/blob/v3.0/dpu/ref_design_docs/README_DPUCZ_Vitis.md#553-known-issues)
When HPC0 or HPC1 port is used to connect the DPU, the HPC hardware cache-coherency mechanism may consume DPU bandwidth either on the AXI bus or in memory, resulting in increased DPU latency on the HPC port. To address this issue, it is suggested to follow the provided recommendation to disable HPC hardware cache-coherency. 
Btw,  the DDR Quality of Service (QoS) is listed as another known issue that can potentially improve the frames per second (FPS) as well

qianglin-xlnx, 2023-07-30:
Closing since no activity for more than 1 month, please reopen if you still have question, thanks

closed 2023-07-30 by qianglin-xlnx