Xilinx/Vitis-AI #1113: The pytorch compilation tool does not support hardsigmoid independent compilation？

jia0511, 2022-12-12:
Hi xilinx,

I replaced the sigmoid operator in efficentNet, but the pytorch version compilation tool prompts that the operator is not supported to be compiled in the DPU, which has greatly affected the model performance, but the tf version compilation tool can support the sigmoid operator independently Compile, I would like to know whether independent compilation is supported in vitis-ai3.0 version?

huisunCompiler, 2023-01-13:
To my knowledge, fix1 + hard-sigmoid + mul + fix2 will be converted to hard-sigmoid-fix op and deployed to DPU if 

- the fix_point of fix2 should be < 7,
- const value = (float) 1.0001220703125

[image]
[image]

I don't know which EfficientNet model you use. We only release EfficientNet-lite_tf2 in vitis-AI 3.0 and your model architecture seems to be EfficientNet-b4 or EfficientNet-b7. Could you please provide the arch of DPU and the xmodel so that we can reproduce the problem.

cabreraam, 2023-01-13:
How are y'all viewing the DPU subgraphs? What is the command? What is this graphical interface?

jia0511, 2023-01-19:
@huisunCompiler 
I don’t know exactly what fix1 and fix2 are. Is there any pytorch example where fix1 + hard-sigmoid + mul + fix2 is transferred to the hard-sigmoid-fix operator? In addition, for example, replacing sigmoid with SE module hardsigmoid in the efficientnet network will affect the final Accuracy?

jia0511, 2023-01-19:
@cabreraam 

I put it directly into xmodel through https://netron.app/ website;

huisunCompiler, 2023-01-20:
@cabreraam 
My netron is a modified version of very old netron. I recommend using the official version https://github.com/lutzroeder/netron, or the website version https://netron.app/.

jia0511, 2023-01-29:
@cabreraam 

Can you answer my concerns problem？

huisunCompiler, 2023-01-31:
@jia0511 ,
Sigmoid can't be transferred to hard-sigmoid. Only hard-sigmoid-fix can be deployed to DPU. Here gives an example of `fix1 + hard-sigmoid + mul + fix2 -> hard_sigmoid-fix` and launch on DPU.

```python
import numpy as np
import xir
import math
np.random.seed(2)
data1 = np.random.randint(low=-8, high=8, size=(1, 14, 14, 240))
const = np.array([1.0001220703125], dtype=np.float32)
g = xir.Graph("h")
data = g.create_op(
    "data1", "data", {"data_type": "FLOAT32", "shape": data1.shape})
input1_fix = g.create_op("input1", "fix", {"bit_width": 8, "fix_point": 2,
                                   "if_signed": True, "round_mode": "DPU_ROUND"}, {"input": [data]})
factor = g.create_const_op("factor", const.astype(np.float32))
hardsigmoid = g.create_op("hardsigmoid", "hard-sigmoid", {}, {"input": [input1_fix]})
mul = g.create_op("mul", "mul", {}, {"input": [hardsigmoid, factor]})
hardsigmoid_fix = g.create_op("hardsigmoid_fix", "fix", {"bit_width": 8, "fix_point": 7,
                                   "if_signed": True, "round_mode": "DPU_ROUND"}, {"input": [mul]})
g.serialize("hard_sigmoid.xmodel")
```
Then, use `vai_c_xir` tool to compile "hard_sigmoid.xmodel", you may find the hard_sigmoid are transfered to hard_sigmoid-fix and deployed on DPU. If it is not the case, could you please provide your lib_info, arch, and xmodel to us? The lib_info can be viewed from

`xir graph your_model_after_compilation.xmodel`

> @huisunCompiler I don’t know exactly what fix1 and fix2 are. Is there any pytorch example where fix1 + hard-sigmoid + mul + fix2 is transferred to the hard-sigmoid-fix operator? In addition, for [...]

jia0511, 2023-02-07:
My DPU type is vck5000 8PE

roycezjq, 2023-04-07:
@jia0511  hi, I'm new to this. How did you replace sigmoid with hard sigmoid? Did you replace in the model definition using torch.nn.HardSigmoid, or during the quantization or optimization?
Thanks

jia0511, 2023-04-07:
Thanks for your reply. I tried to use efficeintNet-b0 model to replace swish with operators, but hard-sigmoid was allocated to CPU during compilation. What is the cause？
[err.log](https://githu
[image]
b.com/Xilinx/Vitis-AI/files/11177102/err.log)