Xilinx/Vitis-AI #1115: Wrong input and output tensor shapes at fpga model execution

konegen, 2022-12-14:
Hi all,

I have a problem executing my .xmodel (TensorFlow 2) on the Xilinx ZCU104 FPGA.
When training my TensorFlow model I define the input as (Batchsize, 8) and my output as (Batchsize, 2). After quantizing and compiling my model to a .xmodel I execute it on my Xilinx ZCU104. Unfortunately, I realized that the input and output tensors just changed dimensions to (Batchsize, 2, 2, 2) for input and (Batchsize, 32) for output.
Like in the mnist example I set the input and output tensors for the FPGA as follows:
```
def runDPU(id,start,dpu,data):
    '''get tensor'''
    inputTensors = dpu.get_input_tensors()
    outputTensors = dpu.get_output_tensors()
    input_ndim = tuple(inputTensors[0].dims)
    output_ndim = tuple(outputTensors[0].dims)
```
So I thought this will set everything automatically after reading the right information from the .xmodel file.
Can anyone help me and tell me what I am doing wrong or what the reasons for this could be?

Thanks

max271828, 2022-12-27:
Hi Konegen,
The dpu could only use the  input which dim>=4, if less than, it will convert it to dim=4.
But I am not clear why the output changed a lot.  Can you share your model to me?

Thanks

qianglin-xlnx, 2023-01-12:
Closing since no activity for more than 2 weeks, please reopen if you still have any questions, thanks

closed 2023-01-12 by qianglin-xlnx